avrdudes/avrdude #131: [bug #22700] Feature Request: Make AVR910 Device Codes more Usable

avrdude-admin, 2021-12-10:
Mike Perks <mikep@oakmicros.com>
Sun 23 Mar 2008 03:34:59 AM UTC

I'm not sure where else to submit this feature request so I'm using a bug for now. There are two items:
1. AVRDude lists the supported devices as returned by the AVR910 programmer and maps these to device names. The last device with the code is listed so for example device code 0x20 is listed as ATtiny84 rather than the more usual AT90S2313. This is perhaps best fixed by listing all the devices that have that device code.
2. AVRDude doesn't recognize the BOOT device codes because they aren't listed in the conf file. They should probably be added to the conf file.

This issue was migrated from https://savannah.nongnu.org/bugs/?22700

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 23 Mar 2008 09:19:51 PM UTC

(It's OK to file a feature request under "Bugs".)

For 1.), I welcome any patches doing so.

For 2.), AVR910 has basically been superseded by AVR109 which
doesn't use device codes anymore.  Do you have any instance
of a bootloader that would really benefit from adding even
more device codes per part since it could not handle the
avr109 implementation?

But again, it would require patches to implement this.

mcuee, 2022-05-30:
@dl8dtl 
This was rather old and since AVR910 has been superseded by AVR109, in that case probably this can be closed.

closed 2022-05-30 by MCUdude