avrdudes/avrdude #147: [bug #25804] configure script does not abort if flex/lex is missing

avrdude-admin, 2021-12-10:
Dan Ståhlberg <None>
Mon 09 Mar 2009 09:28:49 PM UTC

On SUSE 11.1, logged in as root, using BASH-3.2.

Downloaded and extracted AVRDUDE-5.6.

The configure script reports that flex/lex is missing but does not abort. This is easy to miss.

checking dependency style of gcc... gcc3
checking for a BSD-compatible install... /usr/bin/install -c
checking for bison... bison -y
 **checking for flex... no**
 **checking for lex... no**
checking for ranlib... ranlib
checking for tputs in -ltermcap... no
checking for tputs in -lncurses... yes
checking for readline in -lreadline... yes

Running make afterwards gives an error that is hard to figure out:

sed '/^#/ s|y\.tab\.c|config_gram.c|' y.tab.c >config_gram.ct && mv config_gram.ct config_gram.c
rm -f y.tab.c
if gcc -DHAVE_CONFIG_H -I. -I. -I.  -DCONFIG_DIR=\"/usr/local/etc\"  -Wall -g -O2 -MT libavrdude_a-config_gram.o -MD -MP -MF ".deps/libavrdude_a-config_gram.Tpo" -c -o libavrdude_a-config_gram.o `test -f 'config_gram.c' || echo './'`config_gram.c; \
then mv -f ".deps/libavrdude_a-config_gram.Tpo" ".deps/libavrdude_a-config_gram.Po"; else rm -f ".deps/libavrdude_a-config_gram.Tpo"; exit 1; fi
config_gram.y: In function ‘parse_cmdbits’:
config_gram.y:1373: warning: ‘brkt’ may be used uninitialized in this function
:   lexer.l
sed '/^#/ s|\.c|lexer.c|' .c >lexer.c
 **sed: can't read .c: No such file or directory**
 **make[2]:**** [lexer.c] Error 2*
make[2]: Leaving directory `/root/avrdude-5.6'

This issue was migrated from https://savannah.nongnu.org/bugs/?25804

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 02 Jul 2009 09:33:00 AM UTC

Alas, AC_PROG_LEX and AC_PROG_YACC do only check for the
possible alternatives (flex vs. lex, bison vs. byacc vs. yacc),
without any real option to bail out if none of them has been
found.

It's not really easy to extend that check without duplicating
much of the code of these macros.  Ideally, this should be
filed as a feature request to project autoconf so these
macros would optionally allow for code to be run if none of
the tools had been found.

Would you be willing in filing such a feature request?

mariusgreuel, 2022-01-13:
Fixed with CMake project

closed 2022-01-13 by mariusgreuel