avrdudes/avrdude #153: [bug #26345] Feature Request: -S and -Z options from STK500.EXE

avrdude-admin, 2021-12-10:
RE <recycle7777@hotmail.com>
Sun 26 Apr 2009 03:31:39 AM UTC

The STK500 programmer has two useful command line arguments for saving data from the EEPROM during programming before the chip erase.

From the program's help output:

-Z [address]  Loads a value from an EEPROM location prior to chip erase which then can be programmed into FLASH or EEPROM using the S option. Address is a hex value. See appnote AVR053 for more information.

This issue was migrated from https://savannah.nongnu.org/bugs/?26345

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 26 Apr 2009 07:23:41 PM UTC

I'm not very convinced about the usefulness of such an option.

For one, my guess is that this option in stk500.exe simply
predates the days of the EESAVE fuse, where each chip erase
involuntarily erased the EEPROM as well.

Second, why stop at a single byte?  What if someone wants to
read two bytes, or four bytes, or two times two bytes?  Or,
if someone wants to read the oscillator calibration fuse,
and store it automagically in EEPROM?

The same issue came already up with **[bug #17487](https://savannah.nongnu.org/bugs/?17487)**, about how
to handle the oscillator calibration value.

So what I think is needed here would be a fairly generic
scheme "copy this many bytes from this memory space to that
many bytes of that memory space", where each option invokes
a new copy operation.  Unfortunately, this is going to be
a major effort which is beyond my own available capacity.
If someone wants to implement a complete functionality as
outlined here, please speak up on the avrdude-dev mailing
list, file a RFC for the implementation, and then start
implementing it.

MCUdude, 2022-05-29:
I'm closing this because the usefulness is questionable and if something like this would be implemented it should support an arbitrary number of bytes to read and write, not just a single. 

With a little shell scripting, you can also quite easily implement something similar yourself.

```
$ echo "write eeprom 0 0x01 0x02 0x03" | ./avrdude -qq -c usbasp -p atmega644p -t
>>> write eeprom 0 0x01 0x02 0x03 

$ echo "read eeprom 0 0x10" | ./avrdude -qq -c usbasp -p atmega644p -t
>>> read eeprom 0 0x10 
0000  01 02 03 ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
```

The output can obviously be piped into another program such as `awk` to make the output look like you want

```
$ echo "read eeprom 0 0x40" | ./avrdude -qq -c usbasp -p atmega644p -t
>>> read eeprom 0 0x40 
0000  01 02 03 ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

echo "read eeprom 0 0x40" | ./avrdude -qq -c usbasp -p atmega644p -t | awk 'NR>=2{for(i=2;i<=NF-1;i++) printf $i" "; print ""}'
01 02 03 ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
```

Feel free to open a new issue with ideas on ideas on what a more generic implementation would look like!

closed 2022-05-29 by MCUdude