avrdudes/avrdude #163: [bug #27207] very slow with JTAGICE MKII using jtag2isp

avrdude-admin, 2021-12-10:
Markus Schulz <None>
Sun 09 Aug 2009 04:33:37 AM UTC

It takes between 1 to 4 minutes to program the 2kb, sometime I get write errors and I need to restart (I have the same issues with AVRDUDE under Linux and Windows, the Avr Studio (gui and command line) works fine (takes like 5 seconds to program)... please let me know what to send in that gives additional help.

Thanks Markus

PS: here are some screens:

markus@ubuntu:~/avr/HIDKeys linux .2007-03-29_5keys_send$ avrdude -c jtag2isp -p 2313 -P usb -vv

avrdude: Version 5.8, compiled on Aug  7 2009 at 23:01:12
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/usr/local/etc/avrdude.conf"
User configuration file is "/home/markus/.avrduderc"

Using Port                    : usb
Using Programmer              : jtag2isp
avrdude: stk500v2_jtagmkII_open()
avrdude: usbdev_open(): Found JTAGICE mkII, serno: 07000000066E
avrdude: usbdev_open(): using read endpoint 0x82
avrdude: jtagmkII_getsync(): Sending sign-on command: 0x86 (28 bytes msg)
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
boot-loader FW version:        255
firmware version:              6.02
hardware version:              0
S_MCU:
boot-loader FW version:        255
firmware version:              6.02
hardware version:              1
Serial number:                   07:00:00:00:06:6e
Device ID:                       JTAGICEmkII
avrdude: jtagmkII_setparm()
avrdude: jtagmkII_setparm(): Sending set parameter command (parm 0x03, 1 bytes): 0x80 (1 bytes msg)
avrdude: jtagmkII_getsync(): Sending get sync command: 0x80 (1 bytes msg)
AVR Part                      : AT90S2313
Chip Erase delay              : 20000 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom         4    12    64    0 no        128    0      0  4000  9000 0x80 0x7f
flash          4    12   128    0 no       2048    0      0  4000  9000 0x7f 0x7f
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
fuse           0     0     0    0 no          1    0      0     0     0 0x00 0x00
lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00

Programmer Type : JTAGMKII_ISP
Description     : Atmel JTAG ICE mkII in ISP mode
avrdude: jtagmkII_getparm()
avrdude: jtagmkII_getparm(): Sending get parameter command (parm 0x06): 0x81 (3 bytes msg)
Vtarget         : 3.3 V
SCK period      : 1.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9101
avrdude: safemode: Fuse reading not support by programmer.
Safemode disabled.
avrdude: current erase-rewrite cycle count is 2147483647 (if being tracked)
avrdude: jtagmkII_close()
avrdude: jtagmkII_close(): Sending GO command: 0xa4 (1 bytes msg)
avrdude: jtagmkII_close(): bad response to GO command: RSP_ILLEGAL_EMULATOR_MODE
avrdude: jtagmkII_close(): Sending sign-off command: 0x80 (1 bytes msg)

avrdude done.  Thank you.

markus@ubuntu:~/avr/HIDKeys linux .2007-03-29_5keys_send$ make flash
avrdude -s -U HIDKeys.hex -c jtag2isp -p 2313 -P usb -D

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9101
avrdude: current erase-rewrite cycle count is 2147483647 (if being tracked)
avrdude: reading input file "HIDKeys.hex"
avrdude: input file HIDKeys.hex auto detected as Intel Hex
avrdude: writing flash (1928 bytes):

Writing | ################################################## | 100% 225.32s

avrdude: 1928 bytes of flash written
avrdude: verifying flash memory against HIDKeys.hex:
avrdude: load data flash data from input file HIDKeys.hex:
avrdude: input file HIDKeys.hex auto detected as Intel Hex
avrdude: input file HIDKeys.hex contains 1928 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 98.28s

avrdude: verifying ...
avrdude: 1928 bytes of flash verified
avrdude: jtagmkII_close(): bad response to GO command: RSP_ILLEGAL_EMULATOR_MODE

avrdude done.  Thank you.

This issue was migrated from https://savannah.nongnu.org/bugs/?27207

avrdude-admin, 2021-12-10:
Markus Schulz <alpharesearch>
Wed 12 Aug 2009 03:08:01 AM UTC

I just tried the arduino isp and it also is very slow, is this maybe releated to the AT90S2313 and the missing PAGEL and BS2?

Thanks,
Markus

avrdude -s -U HIDKeys.hex -c avrisp -p 2313 -P /dev/ttyUSB0 -b 19200

avrdude: please define PAGEL and BS2 signals in the configuration file for part AT90S2313
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.13s

avrdude: Device signature = 0x1e9101
avrdude: NOTE: FLASH memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: please define PAGEL and BS2 signals in the configuration file for part AT90S2313
avrdude: reading input file "HIDKeys.hex"
avrdude: input file HIDKeys.hex auto detected as Intel Hex
avrdude: writing flash (1928 bytes):

Writing | ################################################## | 100% 245.38s

avrdude: 1928 bytes of flash written
avrdude: verifying flash memory against HIDKeys.hex:
avrdude: load data flash data from input file HIDKeys.hex:
avrdude: input file HIDKeys.hex auto detected as Intel Hex
avrdude: input file HIDKeys.hex contains 1928 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 82.08s

avrdude: verifying ...
avrdude: 1928 bytes of flash verified

avrdude done.  Thank you.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 11 Sep 2013 04:35:02 PM UTC

Can you try this with a recent version, preferrably with the
SVN version (or 6.0rc1)?

avrdude-admin, 2021-12-10:
Matthijs Kooijman <matthijs>
Tue 14 Oct 2014 06:58:33 PM UTC

Isn't this slowness due to the default slow bit clock? With jtag3isp, the default clock is terribly slow, adding `-B 1` to the commandline speeds things up a lot.

mcuee, 2022-05-30:
Not so sure if anyone still has the very old AT90S2313 to test and check if `-B 1` mentioned in the comments by Matthijs helps to speed up the programming or not.

MCUdude, 2022-05-30:
I don't own any AT90S hardware but maybe @dl8dtl does?

mcuee, 2022-07-26:
Since the OP has not come back for a long time and this is related to a very old chip, I will close this as `invalid` for now. Please re-open if an user encountered this issue again.

closed 2022-07-26 by mcuee