avrdudes/avrdude #166: [bug #27501] [Feature Request]: Support AVR32

avrdude-admin, 2021-12-10:
D Hoerl <dhoerl@mac.com>
Tue 22 Sep 2009 08:34:56 PM UTC

There was a prior thread about Atmel not releasing the necessary information to program AVR32, but in fact they did in 2007:

 [http://www.atmel.com/dyn/resources/prod_documents/doc32070.pdf](https://savannah.nongnu.orghttp://www.atmel.com/dyn/resources/prod_documents/doc32070.pdf)

David

PS: yeah, I could try to figure it out. but it would take me forever. I support xlslib on sourceforge, another open source library to write Excel files. There are less than 10 people in the world who could maintain that software now (due to MS complexity) - I'd never ask a user to even try...

This issue was migrated from https://savannah.nongnu.org/bugs/?27501

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 07 Jan 2010 04:25:26 PM UTC

David, can you please update this tracker item?

mcuee, 2022-06-07:
Related issues.
* #735 
* https://github.com/avrdudes/avrdude/issues/239
* https://github.com/avrdudes/avrdude/issues/977

mcuee, 2022-06-23:
AVR32 has been de-focused by Microchip/Atmel long ago. So we will close this issue.

closed 2022-06-23 by mcuee