avrdudes/avrdude #185: [bug #28731] Can't build avrdude 5.10 on OSX with USB support

avrdude-admin, 2021-12-10:
giuliano carlini <giuliano@carlini.com>
Mon 25 Jan 2010 07:22:17 AM UTC

configure script doesn't properly link libusb check program.

It links with "-l libusb", but needs to also link with
"-framework CoreFoundation -framework IOKit".
This is properly if the check script passes, it checks whether
the OS is darwin, and if so adds these to LIBUSB. But also
needs to add them to the check program.

This issue was migrated from https://savannah.nongnu.org/bugs/?28731

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 16 Aug 2011 06:35:10 PM UTC

@avrdude-dev:

Any OS X gurus around to have a look on that?

avrdude-admin, 2021-12-10:
Darell Tan <geekman>
Mon 22 Aug 2011 05:51:33 AM UTC

I've just tried it on my Mac. It compiles fine, but strangely I can remove the CoreFoundation and IOKit framework flags and just leave the -lusb flag and it STILL compiles properly.

My guess is that the gcc I'm using (version 4.0.1 Apple Inc. build 5493) automatically links those frameworks.

If other versions of gcc are to be supported, my suggestion would be to move the $target checks for darwin (line 4646) up to line 4581 where the LIBS are defined, to add the -framework flags for CoreFoundation and IOKit.

What version of gcc are you using?

MCUdude, 2022-01-09:
Issue can probably be closed. I have no issues building Avrdude on my mac.

mariusgreuel, 2022-01-09:
See build instructions for macOS: https://github.com/avrdudes/avrdude/wiki/Building-AVRDUDE-for-macOS

closed 2022-01-09 by mariusgreuel