avrdudes/avrdude #189: [bug #29263] Can't build avrdude on windows using latest cygwin 1.7.1

avrdude-admin, 2021-12-10:
giuliano carlini <None>
Fri 19 Mar 2010 06:23:02 AM UTC

avrdude manual section "A.2 Windows" says that to compile on windows one must use:
./configure LDFLAGS="-static" ...
configure fails. config.log says:
.../ld: cannot find -lgcc_s

Other tweaks to configure and make fail in various ways.

This issue was migrated from https://savannah.nongnu.org/bugs/?29263

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 19 Mar 2010 06:34:19 AM UTC

But: "cannot find -lgcc_s" would be Cygwin's fault...

What happens if you build without -static?

Please elaborate the "various ways".  I don't have Windows
here, I cannot fix anything unless Windows users/developers
provide details.

I know it's less comfortable (as you have to pick all parts
of the puzzle yourself), but you might consider using MinGW
to build.

avrdude-admin, 2021-12-10:
giuliano carlini <gorillacoder>
Fri 19 Mar 2010 07:04:16 AM UTC

I generally work on OSX. I run XP within a VM so that I can support arduino users who use windows. I believe folks who are primarily linux developers can do the same, whether with VirtualBox/KVM/VmWare/or Xen.

I'm porting arduino to the xmega. I've put together packages for OSX. Using a VM I put together a package for Linux users, even though I use Linux infrequently these days. I'm trying to do the same for windows, even though I do not use windows myself.

I spent a couple hours trying to get avrdude to compile on windows yesterday. I don't recall all the failure modes.

I do recall that without the -static, that gcc complains that it no longer support no-cygwin and that I should consider using mingw. Which was somewhat irritating as the avrdude manual says:
To build avrdude from the source You must have
Cygwin ([http://www.cygwin.com/](https://savannah.nongnu.orghttp://www.cygwin.com/)).
As I'm sure you've heard, it takes hours to download and install cygwin. And I tried to get it to work because the manual said that cygwin was required, not mingw. If mingw is the correct build toolchain, it would have been nice if the manual had directed me to use MinGW.

Again, thanks for all the help.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 19 Mar 2010 07:31:30 AM UTC

Well, my current FreeBSD version is too old to run
virtualmachine, I have to upgrade first.  Besides, it
would require a Microsoft license anyway (which I wouldn't
want to obtain, why should I pay for something I don't
really want to use?).

I've heard that Cygwin doesn't want to support MinGW builds
anymore, though I think you could still convince it to do
so by installing an older compiler version.  AVRDUDE Win32
builds have always been MinGW builds technically (in order
to avoid linking against cygwin1.dll, so to not run into the
DLL version mess); the recommendation to build under Cygwin
in MinGW compatibility mode has only been there as Cygwin is
a "one-stop show" with a nice installer, where in MinGW, you
have to pull all the individual packages yourself.

However, if Cygwin could not be convinced to build native
Win32 API (aka. MinGW) binaries anymore, yes, we indeed have
to withdraw the recommendation to use Cygwin as the build
environment from our documentation.

Please note that AVRDUDE didn't change in that respect in
any way, the only changes that are messing you up here are
Cygwin changes.  So if you're interested to keep the ability
for native Win32 API builds in Cygwin in future, you might
complain to the Cygwin maintainers.

avrdude-admin, 2021-12-10:
giuliano carlini <gorillacoder>
Fri 19 Mar 2010 06:29:38 PM UTC

Sorry, I barely have time for arduino. I don't have time to figure out how to get avrdude to compile under windows. For now I'll release xmega arduino on mac and linux and provide a brief explanation about windows.

I understand that it was cygwin that changed. At some level that doesn't really matter to me, I'm stuck. But I appreciate how frustrating it must be for you. You didn't break it, but have to fix it if you want to support avrdude users who are on windows.

As for why should you pay for/obtain windows, well that's a philosophical issue. I'd say that because doing so furthers your ideological bent. A lot of folks move to linux/bsd because they gain experience with how great open software is through cygwin or other windows ports of great open software. If you purchase windows, and that results in just two folks moving away from windows, you've won.

I think avrdude is great. If you like, I'll pop for a copy of windows for you. Or if you prefer, I'll send you the disks to an old copy of windows I'm not using, so that MS gets no more money out of it. If you are interested, just let me know which you prefer. If you are not interested, I can respect that.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 19 Mar 2010 09:36:23 PM UTC

I've removed the recommendation for Cygwin, mentioned that
starting with Cygwin 1.7 plain Win32 API builds are no longer
supported (the way they used to).  I also mentioned MinGW as
an alternative build environment.  However, I don't really
grok MinGW's packaging details, so I cannot recommend the
packages that have to be installed.

Assigning it to Eric in the hope he might fill in the gap
in the documentation about MinGW.

MCUdude, 2022-01-09:
Issue can probably be closed

mariusgreuel, 2022-01-09:
cmake works for all current build systems.

closed 2022-01-09 by mariusgreuel