avrdudes/avrdude #190: [bug #29264] avrdude upload to xmega using stk500 protocol using byte addresses rather than word addresses

avrdude-admin, 2021-12-10:
giuliano carlini <None>
Fri 19 Mar 2010 06:26:58 AM UTC

STK500 'U" command takes a word address. Indeed, avrdude seems to be passing word addresses to the arduino bootloader for atmega chips. However, avrdude seems to be passing byte addresses for xmega28a1.

[file #19981: xmegaTest0.pde](https://savannah.nongnu.org/bugs/download.php?file_id=19981)
[file #19982: avrdude-29264.log](https://savannah.nongnu.org/bugs/download.php?file_id=19982)

This issue was migrated from https://savannah.nongnu.org/bugs/?29264

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 19 Mar 2010 06:35:17 AM UTC

Could you please attach a communication trace file?  (Output
from avrdude -vvvv, captured in a text file.)

avrdude-admin, 2021-12-10:
giuliano carlini <gorillacoder>
Fri 19 Mar 2010 07:14:05 AM UTC

attached are the input arduino sketch and the resulting avrdude trace.

Thanks.

(**[file #19981](https://savannah.nongnu.org/support/download.php?file_id=19981)**, **[file #19982](https://savannah.nongnu.org/support/download.php?file_id=19982)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 19 Mar 2010 07:58:15 AM UTC

Thanks for the logfile.  I just realized Arduino is still using
STK500v1 (it's probably the last consumer of that protocol :).  I
don't think anybody did anything lately here (except for fixing the
Arduino reset), so if it works with Xmegas at all, it's purely as a
side-effect from other changes introduced to the generic code.

Patches would be very welcome.  I think I've got some STK500v1
firmware still lying around somewhere, so I might be able to test it
on a real STK500 for comparison.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 19 Apr 2012 01:45:02 PM UTC

Sorry, STK500v1 is not a priority to me right now.

Again, if anyone wants to supply patches, they're welcome.

avrdude-admin, 2021-12-10:
Michael Dreher <nospam2000>
Sun 10 Mar 2013 08:12:45 PM UTC

When the opcodes AVR_OP_LOADPAGE_LO and AVR_OP_READ_LO are not defined in avrdude.conf for the mem type flash for this device, it uses a_div=1 (byte addressing) instead of a_div=2 (word addressing). So this be easily fixed in avrdude.conf.

From stk500.c stk500_paged_write():
if ((m->op[AVR_OP_LOADPAGE_LO]) || (m->op[AVR_OP_READ_LO]))
a_div = 2;
else
a_div = 1;

Copy and adapt the definitions from ATmega2560.

mcuee, 2022-06-05:
This seems to be related to the stk500v1 EEPROM discussion here. It could be the real STK500 v1 is correct and just that the Arduino bootloader is not following the real STK500 v1 in this aspect.
https://github.com/avrdudes/avrdude/issues/967

But it affects xmega and I can not test that.

mcuee, 2022-06-20:
> When the opcodes AVR_OP_LOADPAGE_LO and AVR_OP_READ_LO are not defined in avrdude.conf for the mem type flash for this device, it uses a_div=1 (byte addressing) instead of a_div=2 (word addressing). So this be easily fixed in avrdude.conf.

AVR_OP_LOADPAGE_LO and AVR_OP_READ_LO  do not seem to exist in avrdude.conf now.

mcuee, 2022-07-10:
Since this is related to STK500v1 bootloader, I am not so sure if this is really of importance for ATxmega or not. The more popular bootloader seems to using FLIPv2.

But there seem to be bootloaders based on arduino bootloader (stk500v1) here for xmega.
https://github.com/XMegaForArduino/arduino/tree/master/bootloaders

mcuee, 2022-07-10:
@dl8dtl 
I belive the STK500v1 FW does not really support ATxmega. So I am not so sure how you can test this out with a real v1 FW for the STK500.

In that case, this seems to be an `enhancement` to me.

mcuee, 2022-08-27:
Based on the comments here, I think the issue is not valid any more. 
https://github.com/XMegaForArduino/arduino/tree/master/bootloaders/xmega128a1

mcuee, 2022-08-27:
I will close this issue for now. Please re-open if the issue pops up again.

closed 2022-08-27 by mcuee