avrdudes/avrdude #200: [bug #29896] Add support for XMEGA specific AVR109 protocol extensions used by XBoot

avrdude-admin, 2021-12-10:
Alex Forencich <None>
Sun 16 May 2010 07:42:50 PM UTC

The AVR109 butterfly protocol does not support access to the user signature row and the production signature row on XMega devices.  The XBoot bootloader adds support for these locations by adding the memory types "U" and "P" in addition to the existing "E" and "F" types.

Updating AVRDUDE to support this would be trivial - it involves adding a couple of additional conditions to the block read and write fuctions in butterfly.c.  The only question is whether or not to add them universally and check for error conditions or to only support them for specific bootloaders.

This issue was migrated from https://savannah.nongnu.org/bugs/?29896

mcuee, 2022-06-05:
I guess xmega is not popular that this has never been implemented.

mcuee, 2022-06-07:
Since this involves user signature row, I am not so sure if the following two issues are related as well.
* #413
* #379

mcuee, 2022-08-07:
@alexforencich
Not so sure if you are still interested in this topic. If yes, please help to come out the pull request. Thanks.

MCUdude, 2022-09-26:
@alexforencich it would be great if you had time to look at this! I have an ATxmega256A3BU board I can use to help out with testing. Since this is an Xboot-exclusive feature(isn't it?), maybe we can use the detected bootloader type to enable these features.

```
$ ./avrdude -cavr109 -p atxmega256a3bu -b 115200 -P /dev/cu.usbserial-1410

Connecting to programmer: .
Found programmer: Id = "XBoot++"; type = S
    Software Version = 1.7; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=512 bytes.

Programmer supports the following devices:
    Device code: 0x7b
````

PR #1108 (2022-10-05): AVR109: support for prodsig read and usersig read/write
This PR allows the AVR109 based [Xboot bootloader](https://github.com/alexforencich/xboot/) to read `prodsig` and read/write to the `usersig` memory on any Xmega that Xboot supports. Tested on an ATxmega256A3BU.

This is an extension of the original AVR109 protocol, and I've not added any guards to prevent users from reading/writing to thee memories. Mostly because (I think) it's unlikely that Xmega users are running a different avr109 bootloader implementation.

Closes #200
closed 2022-10-05 by stefanrueger