avrdudes/avrdude #207: [bug #30498] Add support for new avr devices (atmega324pa)

avrdude-admin, 2021-12-10:
rr <logicnibble@gmail.com>
Tue 20 Jul 2010 04:28:04 PM UTC

Some new devices are not supported by avrdude, like the atmega324 variants: atmega324a and atmega324pa (only atmega324p is supported).
I've tested with version 5.10.
I have to use the -F option to force the programming in atmega324p mode. I guess it works fine, but that needs to be changed.
Please check all other devices and different signatures that are not supported.

[file #23837: atmegaXX4.conf](https://savannah.nongnu.org/bugs/download.php?file_id=23837)
[file #23844: atmegaXX4_corrected.conf](https://savannah.nongnu.org/bugs/download.php?file_id=23844)

This issue was migrated from https://savannah.nongnu.org/bugs/?30498

avrdude-admin, 2021-12-10:
jonathan nifenecker <crazyiop>
Fri 19 Aug 2011 09:45:40 AM UTC

part definition for :
atmega164a
atmega164pa
atmega324a
atmega644a
atmega644pa
atmega1284
made from xml and datasheet.

atmega324pa and atmega1284p already done in latest .conf file.

(**[file #23837](https://savannah.nongnu.org/support/download.php?file_id=23837)**)

avrdude-admin, 2021-12-10:
jonathan nifenecker <crazyiop>
Sun 21 Aug 2011 09:59:42 PM UTC

just fix the error (a mistake in pgm_enable command)

(**[file #23844](https://savannah.nongnu.org/support/download.php?file_id=23844)**)

MCUdude, 2022-01-07:
@dl8dtl this issue can be closed

closed 2022-01-07 by dl8dtl