avrdudes/avrdude #211: [bug #30736] Patch to stk500v2 to support non-USB XPROG

avrdude-admin, 2021-12-10:
Erik Walthinsen <omega@omegacs.net>
Tue 10 Aug 2010 06:59:42 PM UTC

In developing a stk500v2 XPROG bootloader for my product, I ran into a problem with avrdude switching into the USB-only unframed version of the protocol after SIGN_ON.  It does so because of an assumption that every stk500v2-xprog device will be communicating over USB.  In my case I'm running over a serial link (and soon running that over a mutliplexed protocol), and the assumption is no longer correct.

I did a quick hack to the stk500v2.c code in order to make both cases work.  Basically it adds a struct var 'bareproto', which is used to decide whether to send the stk500 commands framed or unframed, rather than the specific programmer type.  The flag itself is set when the actual comms port is selected, and currently is only set if it's USB.

Being a quick hack I don't expect this patch to go anywhere itself.  However, something like it should be put in place, since non-USB XPROG implementations are likely to show up (especially since I'll be releasing mine at some point soon...).

Likely a better way to implement it would be an avrdude.conf flag that would specifically set it to "framed-mode".  Along those lines it would be nice to have some way to define programmers by their SIGN_ON responses.  Since SIGN_ON is always sent in full stk500 framed mode no matter what, having avrdude check through the .conf for a string match (rather than hard-coded) and then picking up the "framed-mode" flag before continuing would make it easy to define new programmer types for bootloaders, etc.

Due to my development schedule I'm unable to do anything further on this at this point, but if it's still an issue when I have a bit more time, I'll see if I can give it a go.

[file #21184: stk500v2-bareproto.diff](https://savannah.nongnu.org/bugs/download.php?file_id=21184)

This issue was migrated from https://savannah.nongnu.org/bugs/?30736

mcuee, 2022-05-27:
Is this still relevant now? 

In any case, I think this issue can be closed as the author is not going to continue the work.

MCUdude, 2022-05-27:
I agree!

closed 2022-05-27 by MCUdude