avrdudes/avrdude #221: [bug #32062] Enhancement: Validate params before erasing chip

avrdude-admin, 2021-12-10:
rmann <None>
Wed 05 Jan 2011 10:49:04 PM UTC

I notice that avrdude erases the chip before programming it. If you make a mistake (the one I made was to mis-type the command to read fuse settings), it will erase the chip, then fail to open the programming file (that I mis-specified as mentioned).

It would be nice if it validated the input first, for example, by trying to open the specified file first, then proceeding to erase before programming. Stuff like that.

This issue was migrated from https://savannah.nongnu.org/bugs/?32062

mcuee, 2022-05-30:
Looks like a nice enhancement. On the other hand, the current code already warns about the chip erase sufficiently.

mcuee, 2022-06-07:
@dl8dtl and @MCUdude 
Probably this can be closed as not planned. Of course if someone comes with a PR it can be re-opened.

MCUdude, 2022-06-08:
I agree! A PR is always welcome though.

closed 2022-06-08 by MCUdude