avrdudes/avrdude #226: [bug #33160] JTAGICE mkII in JTAG mode, target not started correctly

avrdude-admin, 2021-12-10:
Stefan May <smay@4finger.net>
Mon 25 Apr 2011 09:32:15 PM UTC

After programming with the JTAGICE mkII in JTAG mode, the target is not started correctly. The CMND_RESET will change the target to STOPPED state according to AVR067. The target is then started by AvrDude with CMND_GO. In my case all commands get a successful response, but the target will not start.

I played around a little and replaced the CMND_RESET with CMND_RESTORE_TARGET (0x23). After this change the target started as expected.

[file #27710: cmnd_restore.patch](https://savannah.nongnu.org/bugs/download.php?file_id=27710)

This issue was migrated from https://savannah.nongnu.org/bugs/?33160

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 16 Aug 2011 03:04:51 PM UTC

Can you explain your environment a little more?

. which controller type?
. which make of the JTAGICE mkII (original Atmel, Clone, ...)?
. which firmware version?
. does it only happen after programming the flash, or even
when just running the avrdude command without any -U
option (where it is supposed to just issue one reset)?

So far, nobody else appears to have observed this behaviour.

avrdude-admin, 2021-12-10:
Damien Espitallier <espitall>
Fri 03 Feb 2012 09:33:44 AM UTC

I have the same issue with ATXmega128A1 (and original atmel JTAGICE mkII)

I only test the unpatched avrdude with 6.06 firwmare but the correction (use of CMND_RESTORE_TARGET) works fine with 6.06 and 7.13

MCUdude, 2022-06-26:
I suspect #979 will resolve this issue as well

mcuee, 2022-06-27:
Let's close this as well since #979 has been merged. If people encounter this issue, they can reopen.

closed 2022-06-27 by mcuee