avrdudes/avrdude #228: [bug #33484] wishlist: halfkay (teensyduino) support

avrdude-admin, 2021-12-10:
chrysn <chrysn@fsfe.org>
Mon 06 Jun 2011 12:35:20 PM UTC

several arduino-style devices called 'teensyduino'[1] are not shipped
with the arduino typical serial programming, but with their own
bootloader protocol called halfkay[2]. they ship an own command line
client[3], which is not included in debian (i have created a minimal
package at [4], but unless i'm convinced that this tool is the right way
to go, i don't plan on supporting it for inclusion in debian).

the halfkay protocol[2] itself is quite stu^H^Himple -- no reading, no
fuses, no identification, writing only in 128byte blocks.

if it is not too simple for avrdude to support, i'd suggest to include
this protocol in avrdude. (for matters of being too simple, i'd like to
point out that other programmers are only partial as well, for example
the arduino bootloader reports fuses as 0x00 and doesn't accept writes.)
support for halfkay would mean that new teensyduino users could get into
it much faster (using well-known tools).

(for reference, the bug was originally filed in debian as #629360[5])

[1] [http://www.pjrc.com/teensy/teensyduino.html](https://savannah.nongnu.orghttp://www.pjrc.com/teensy/teensyduino.html)
[2] [http://www.pjrc.com/teensy/halfkay_protocol.html](https://savannah.nongnu.orghttp://www.pjrc.com/teensy/halfkay_protocol.html)
[3] [http://www.pjrc.com/teensy/loader_cli.html](https://savannah.nongnu.orghttp://www.pjrc.com/teensy/loader_cli.html)
[4] [http://archive.amsuess.com/pool/main/t/teensy-loader-cli/](https://savannah.nongnu.orghttp://archive.amsuess.com/pool/main/t/teensy-loader-cli/)
[5] [http://bugs.debian.org/629360](https://savannah.nongnu.orghttp://bugs.debian.org/629360)

This issue was migrated from https://savannah.nongnu.org/bugs/?33484

avrdude-admin, 2021-12-10:
Marcin Cieślak <saper>
Mon 27 Jun 2011 09:06:22 AM UTC

What's interesting, the original halfkey bootloader code (512 bytes) seems to be proprietary and protected by the lock bits.

There seems to be an open source replacement, however:

[http://elasticsheep.com/2010/01/installing-the-lufa-teensyhid-bootloader/](https://savannah.nongnu.orghttp://elasticsheep.com/2010/01/installing-the-lufa-teensyhid-bootloader/)

avrdude-admin, 2021-12-10:
Roger Wolff <rewolff>
Sun 05 Feb 2012 09:08:14 PM UTC

The replacement LUFA bootloader takes a whopping 4k of your flash.

On the other hand, that's the same as with the default DFU bootloader.

mariusgreuel, 2022-01-12:
Added in #802.

closed 2022-01-12 by mariusgreuel