avrdudes/avrdude #231: [bug #34271] avrdude only writes the eeprom in 8-byte steps (stk500v2)

avrdude-admin, 2021-12-10:
Niko D <niko.delarich@gmail.com>
Sun 11 Sep 2011 10:25:24 PM UTC

When uploading eeprom data to an at90can32 over ISP (STK500v2 bootloader), avrdude only transfers data correctly if the size is a multiple of 8.

If the data size is e.g. 12, only the first 8 bytes are transferred. No problems occur when transferring 8bytes or 16bytes. I verified that this is not a problem with my application code by running the command in verbose mode, where it is clearly visible that avrdude sends 8 as the size if 8 < size <16.

My setup:
at90can32 with a modified version of Peter Fleury's STK500V2 bootloader
avrdude v5.11 (latest).
I also tried it with an older version, 5.8cvs, yielding the same result.

This issue was migrated from https://savannah.nongnu.org/bugs/?34271

mcuee, 2022-06-03:
Not so sure if this is related. The stk500v2 bootloader implementation out there seems to have problem with EEPROM Read. So probably it has problem with EEPROM write as well. 
* https://github.com/avrdudes/avrdude/issues/231

MCUdude, 2022-06-03:
Agree. OP didn't provide any verbose Avrdude output either...

closed 2022-06-03 by MCUdude

mcuee, 2022-06-04:
Not so sure if this is relevant or not.
https://www.eevblog.com/forum/microcontrollers/bootloader-woes-corruption-of-data-being-flashed/

MCUdude, 2022-06-04:
I'm currently on vacation, and only brought my phone. Is this basically just the stk500v2 bootloader EEPROM bug again?

mcuee, 2022-06-04:
Yes looks like a problem again for the bootloader.
I have the EEPROM write problem as well with the wiring bootloader even though the symptom is not exactly the same as the original reported issue.
https://github.com/avrdudes/avrdude/issues/515#issuecomment-1146535985

mcuee, 2022-06-04:
The code changelog says the EEPROM write function has not been tested.
https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/stk500v2/stk500boot.c#L87
```
//*	Jan  1,	2012	<MLS> Issue 543: Write EEPROM now does something (NOT TESTED)
```

mcuee, 2022-06-04:
It seems to write only the first 8 bytes of EEPROM, worse than reading.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 0x00 1 2 3" | ./avrdude -c wiring -P COM6 -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 0x00 1 2 3

Writing | ################################################## | 100% 0.81s

avrdude>
avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.82s

0000  01 02 03 04 05 06 07 08  ff ff ff ff ff ff ff 10  |................|
0010  ff ff ff 03 ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-04:
Comparison for EEPROM reading.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 0x00 1 2 3" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> write eeprom 0x00 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 16 0x00 1 2 3

Writing | ################################################## | 100% 4.05s

avrdude>
avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.82s

0000  01 02 03 04 05 06 07 08  09 0a 0b 0c 0d 0e 0f 10  |........     ...|
0010  00 01 02 03 ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c wiring -P COM6 -p atmega2560 -t
avrdude.exe: stk500v2_ReceiveMessage(): timeout

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.07s

0000  01 02 03 04 05 06 07 08  00 01 02 03 ff ff ff ff  |................|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-04:
But the conclusion here is different from my experiences. It is said that EEPROM write should work. Maybe someone else can try to see if the bootloader works for EEPROM writing or not.

Ref: https://forum.arduino.cc/t/trouble-reading-writing-eeprom-via-avrdude/467537

MCUdude, 2022-06-04:
Yes, I did a little testing a while back, and my conclusion was that stk500v2 EEPROM writing works, but reading fails. I used the bootloader binary bundled with Arduino IDE.

mcuee, 2022-06-04:
Maybe my clone (with CH240 using 12MHz crystal, and ATmega2560 using cheap 16MHz resonator) got some problem with the official bootloader. 

I will stop using this board for testing and get another Mega2560 board to see if that helps.

MCUdude, 2022-06-04:
Have you tried to re-flash the bootloader?

mcuee, 2022-06-04:
> Have you tried to re-flash the bootloader?

Yes I have done that with the following bootloader hex file.
https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/stk500v2/Mega2560-prod-firmware-2011-06-29.hex

Arduino Sketech Download is also a bit flaky for the Mega2560 CH340 Clone -- I need to follow the trick to reset the Mega2560 quickly to get the download to be successful, if not timeout error will happen.
Ref: https://youtu.be/tAzjO4v7oF4

mcuee, 2022-06-04:
Without the reset trick -- run log within Arduino
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\xiaof\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM6
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Wiring
         Description     : Wiring
         Programmer Model: AVRISP
         Hardware Version: 15
         Firmware Version Master : 2.10
         Vtarget         : 0.0 V
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9801 (probably m2560)
avrdude: reading input file "C:\Users\xiaof\AppData\Local\Temp\arduino_build_365699/Blink.ino.hex"
avrdude: writing flash (1536 bytes):

Writing | ########avrdude: stk500v2_recv(): checksum error
avrdude: stk500v2_ReceiveMessage(): timeout
avrdude: stk500v2_ReceiveMessage(): timeout
########################################## | 100% 14.45s

avrdude: 1536 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Temp\arduino_build_365699/Blink.ino.hex:
avrdude: load data flash data from input file C:\Users\xiaof\AppData\Local\Temp\arduino_build_365699/Blink.ino.hex:
avrdude: input file C:\Users\xiaof\AppData\Local\Temp\arduino_build_365699/Blink.ino.hex contains 1536 bytes
avrdude: reading on-chip flash data:

Reading | #########################the selected serial port Reading | ######################### does not exist or your board is not connected
######################### | 100% 0.18s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0c
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```

mcuee, 2022-06-05:
Same for avrdude.

Without the reset trick it will not work.
```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> .\avrdude.exe -p m2560 -c wiring -P COM6 -D -U .\Blink.ino.mega.hex -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : wiring
             AVR Part                      : ATmega2560
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom                 65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
               flash                  65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
               lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
               calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

             Programmer Type : Wiring
             Description     : Wiring
             Programmer Model: AVRISP
             Hardware Version: 15
             Firmware Version Master : 2.10
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\Blink.ino.mega.hex"
avrdude.exe: input file .\Blink.ino.mega.hex auto detected as Intel Hex
avrdude.exe: writing flash (1536 bytes):

Writing | #########################                          | 50% 0.12savrdude.exe: stk500v2_recv(): checksum error
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_ReceiveMessage(): timeout
avrdude.exe: stk500v2_getsync(): timeout communicating with programmer
avrdude.exe: stk500v2_ReceiveMessage(): timeout
...
```
[... 110 lines cut ...]
             Vtarget         : 0.0 V
             SCK period      : 0.1 us

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: reading input file ".\Blink.ino.mega.hex"
avrdude.exe: input file .\Blink.ino.mega.hex auto detected as Intel Hex
avrdude.exe: writing flash (1536 bytes):

Writing | ################################################## | 100% 0.24s

avrdude.exe: 1536 bytes of flash written
avrdude.exe: verifying flash memory against .\Blink.ino.mega.hex:
avrdude.exe: input file .\Blink.ino.mega.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.18s

avrdude.exe: 1536 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-06-05:
@MCUdude Somehow the bootloader in your repo does not work for this board. Maybe I need to use another one.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> ./avrdude -c dragon_isp -p atmega2560 -U .\optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
             To disable this feature, specify the -D option.
avrdude.exe: erasing chip
avrdude.exe: reading input file ".\optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex"
avrdude.exe: input file .\optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex auto detected as Intel Hex
avrdude.exe: writing flash (262144 bytes):

Writing | ################################################## | 100% 0.00s

avrdude.exe: 262144 bytes of flash written
avrdude.exe: verifying flash memory against .\optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex:
avrdude.exe: input file .\optiboot_flash_atmega2560_UART0_115200_16000000L_B7_BIGBOOT.hex auto detected as Intel Hex

Reading | ################################################## | 100% 0.00s

avrdude.exe: 262144 bytes of flash verified

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> ./avrdude -c arduino -P COM6 -p atmega2560 -t
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> ./avrdude -c arduino -P COM6 -p atmega2560 -v

avrdude.exe: Version 7.0
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is "C:/work/avr/avrdude_test/avrdude-v7.0-windows-x64/avrdude.conf"

             Using Port                    : COM6
             Using Programmer              : arduino
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
...

```

mcuee, 2022-06-05:
Erasing the chip (including EEPROM and the reflash the booloader, here is the result.

Now the read and write seem to match. But using AVR Dragon shows that the write is not that correct.

```
PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c wiring -P COM6 -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.08s

0000  01 02 03 04 05 06 07 08  09 0a 0b 0c ff ff ff ff  |........    ....|
0010  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude-v7.0-windows-x64> echo "read eeprom 0 0x40" | ./avrdude -c dragon_isp -p atmega2560 -t

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude.exe: Device signature = 0x1e9801 (probably m2560)
avrdude> >>> read eeprom 0 0x40

Reading | ################################################## | 100% 0.82s

0000  01 02 03 04 05 06 07 08  ff ff ff ff ff ff ff ff  |................|
0010  09 0a 0b 0c ff ff ff ff  ff ff ff ff ff ff ff ff  |    ............|
0020  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|
0030  ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff  |................|

avrdude>
avrdude.exe done.  Thank you.

```

mcuee, 2022-06-05:
I will create a new issue for the EEPROM problem.

mcuee, 2022-06-05:
> Yes, I did a little testing a while back, and my conclusion was that stk500v2 EEPROM writing works, but reading fails. I used the bootloader binary bundled with Arduino IDE.

@MCUdude Yes you are right. EEPROM write is good with the more proper bootloader hex file.
https://github.com/avrdudes/avrdude/issues/986#issuecomment-1146736368

I should not use this one.
https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/stk500v2/Mega2560-prod-firmware-2011-06-29.hex

I should use this one which works better (but still with the EEPROM read problem).
https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/stk500v2/stk500boot_v2_mega2560.hex