avrdudes/avrdude #236: [bug #34436] giveio does not work on windows 7 64-bit.

avrdude-admin, 2021-12-10:
Rutr <artuczapl@gmail.com>
Thu 29 Sep 2011 07:27:33 PM UTC

giveio does not work on windows 7 64-bit.

This issue was migrated from https://savannah.nongnu.org/bugs/?34436

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 30 Sep 2011 04:56:21 AM UTC

giveio is not maintained by the avrdude project; we merely
add it as a convenience to those users who find it helpful.

Ideally, all operating systems would ship with a parallel
port driver that doesn't require a hack like giveio.  So far,
only Linux, Solaris and FreeBSD do that though.  If anyone
comes up with a Windows counterpart and a patch for using
it, we are willing to integrate that method.  Otherwise, we
have to close this as "won't fix" (rather: "can't fix")
some day.

avrdude-admin, 2021-12-10:
Jason Hecker <roklobster>
Fri 24 Feb 2012 02:57:37 PM UTC

Is there any reason to not have inpout32 from [http://www.highrez.co.uk/](https://savannah.nongnu.orghttp://www.highrez.co.uk/) made to work with ppiwin.c?  I'd do it except I have no parallel port any more.

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Mon 09 Nov 2015 09:12:31 PM UTC

Maybe this giveio replacement will work, [http://www.codeproject.com/Tips/985807/Enable-I-O-Access-From-User-Mode](https://savannah.nongnu.orghttp://www.codeproject.com/Tips/985807/Enable-I-O-Access-From-User-Mode)

mariusgreuel, 2022-01-12:
giveio.sys should no longer be used. Please use a USB programmer.

closed 2022-01-12 by mariusgreuel