avrdudes/avrdude #239: [bug #34767] Avrdude to enable AVR Dragon to program AVR32 parts

avrdude-admin, 2021-12-10:
Julien Aube <None>
Mon 07 Nov 2011 08:49:21 AM UTC

Attached is a patch to let the AVR-Dragon program the avr32 parts.

The AVR-Dragon uses the same protocol than the jtagmkII probe, so the patch apply the same code for the Dragon.

It enable a new probe : dragon_avr32 which enable AVR32-specific calls, like the jtagmkII probe does.

This patch modifies the jtagmkII.c/.h files as well as the lex/yacc files, and the configuration example.

[file #24317: dragon_avr32.patch](https://savannah.nongnu.org/bugs/download.php?file_id=24317)
[file #24319: dragon_avr32_v2.patch](https://savannah.nongnu.org/bugs/download.php?file_id=24319)

This issue was migrated from https://savannah.nongnu.org/bugs/?34767

avrdude-admin, 2021-12-10:
Julien Aube <jaube>
Mon 07 Nov 2011 09:31:22 AM UTC

Added dragon_avr32_v2.patch which:

- Add the dragon_avr32 item in the avrdude.conf.in config file template

- Fix a bug in jtagmkII_initialize32(): call jtagmkII_set_devdescr() before using the flash_pagesize variable, since this variable value is set into jtagmkII_set_devdescr().

The consequence of this bug was a malloc(0) on jtagmkII.c:2948, and then any read/write on the page was crashing.

Now the <dump> command works on a UC3A1256.

(**[file #24319](https://savannah.nongnu.org/support/download.php?file_id=24319)**)

mcuee, 2022-05-24:
I only have one UC3 device which is UC3A3256 (UC3-XPlained board from Microchip/Atmel). Avr Dragon works fine under Microchip Studio.

Not so sure if the above patch works for my device or not. I will try this patch later.

mcuee, 2022-06-23:
AVR32 has been de-focused by Microchip/Atmel long ago. So we will close this issue.

closed 2022-06-23 by mcuee

mcuee, 2023-06-11:
Patch file:
```
Index: jtagmkII.c
===================================================================
--- jtagmkII.c	(révision 1018)
+++ jtagmkII.c	(copie de travail)
@@ -166,6 +166,9 @@
                                  unsigned int page_size,
                                  unsigned int addr, unsigned int n_bytes);
 
+static int jtagmkII_dragon_open32(PROGRAMMER * pgm, char * port);
+
+
 void jtagmkII_setup(PROGRAMMER * pgm)
 {
   if ((pgm->cookie = malloc(sizeof(struct pdata))) == 0) {
@@ -1676,7 +1679,81 @@
   return 0;
 }
 
+static int jtagmkII_dragon_open32(PROGRAMMER * pgm, char * port)
+{
+  int status;
+  unsigned char buf[6], *resp;
+  long baud;
 
+  if (verbose >= 2)
+    fprintf(stderr, "%s: jtagmkII_dragon_open()\n", progname);
+
+  /*
+   * The JTAG ICE mkII always starts with a baud rate of 19200 Bd upon
+   * attaching.  If the config file or command-line parameters specify
+   * a higher baud rate, we switch to it later on, after establishing
+   * the connection with the ICE.
+   */
+  baud = 19200;
+
+  /*
+   * If the port name starts with "usb", divert the serial routines
+   * to the USB ones.  The serial_open() function for USB overrides
+   * the meaning of the "baud" parameter to be the USB device ID to
+   * search for.
+   */
+  if (strncmp(port, "usb", 3) == 0) {
+#if defined(HAVE_LIBUSB)
+    serdev = &usb_serdev;
+    baud = USB_DEVICE_AVRDRAGON;
+#else
+    fprintf(stderr, "avrdude was compiled without usb support.\n");
+    return -1;
+#endif
+  }
+
+  strcpy(pgm->port, port);
+  if (serial_open(port, baud, &pgm->fd)==-1) {
+    return -1;
+  }
+
+  /*
+   * drain any extraneous input
+   */
+  jtagmkII_drain(pgm, 0);
+
+  if (jtagmkII_getsync(pgm, EMULATOR_MODE_JTAG) < 0)
+    return -1;
+
+  // AVR32 "special"
+  buf[0] = CMND_SET_PARAMETER;
+  buf[1] = 0x2D;
+  buf[2] = 0x03;
[... 135 lines cut ...]
 programmer
+  id    = "dragon_avr32";
+  desc  = "Atmel AVR Dragon in AVR32 mode";
+  baudrate = 115200;
+  type  = dragon_avr32;
+;
+
+programmer
   id    = "pavr";
   desc  = "Jason Kyle's pAVR Serial Programmer";
   type  = avr910;
Index: lexer.l
===================================================================
--- lexer.l	(révision 1018)
+++ lexer.l	(copie de travail)
@@ -144,6 +144,7 @@
 dragon_jtag      { yylval=NULL; return K_DRAGON_JTAG; }
 dragon_pdi       { yylval=NULL; return K_DRAGON_PDI; }
 dragon_pp        { yylval=NULL; return K_DRAGON_PP; }
+dragon_avr32     { yylval=NULL; return K_DRAGON_AVR32; }
 eecr             { yylval=NULL; return K_EECR; }
 eeprom           { yylval=NULL; return K_EEPROM; }
 enablepageprogramming { yylval=NULL; return K_ENABLEPAGEPROGRAMMING; }

```