avrdudes/avrdude #243: [bug #34994] avrdude does not send PDI offsets when erasing XMEGA chip

avrdude-admin, 2021-12-10:
John Voltz <john.voltz@gmail.com>
Mon 05 Dec 2011 06:59:04 PM UTC

avrdude 5.11.1 does not erase ATxmega16A4.

[file #24540: avrdude-5.11.1-xmega-set-PDI-offsets.patch](https://savannah.nongnu.org/bugs/download.php?file_id=24540)
[file #24539: xmega_erase_avrstudio_4_19.log](https://savannah.nongnu.org/bugs/download.php?file_id=24539)

This issue was migrated from https://savannah.nongnu.org/bugs/?34994

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 02 Sep 2013 06:22:41 PM UTC

Pleae test again with SVN trunk, or (if you cannot do that)
with the publically available 6.0rc1 tarball.

Many things have changed between 5.11.1 and trunk, and it's
more than likely this one is fixed now.

mcuee, 2022-07-12:
I will close this issue for now.

closed 2022-07-12 by mcuee