avrdudes/avrdude #248: [bug #35295] Avrdude detects errors in outputfile "late".

avrdude-admin, 2021-12-10:
rewolff <None>
Thu 12 Jan 2012 03:20:46 PM UTC

avrdude will spend time reading the whole chip before trying to open the output file while performing a "read chip" operation.

I had read the chip "as root" and so the file existed and wasn't writable by me (I thought I had specified a new filename, but hadn't). So I spent a minute watching slow # signs before it reported: can't open output file. :-(

This issue was migrated from https://savannah.nongnu.org/bugs/?35295

avrdude-admin, 2021-12-10:
Roger Wolff <rewolff>
Thu 12 Jan 2012 03:21:40 PM UTC

The same holds for "forgetting to specify the output format". Then too the error comes AFTER the time-consuming step of reading the chip.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 11 Sep 2013 09:22:48 AM UTC

Fixing this requires some major restructuring the way the
"update" operation is implemented, alas.  Lowering the
priority, as I simply don't have the time for this right
now.

mcuee, 2022-05-29:
Since this has been lower priority for so many years and it is more like dealing with a user erorr use cases, maybe just a documentation fix will do, telling the user to check in case such error happens.

mcuee, 2022-06-06:
@MCUdude or @dl8dtl Maybe this can be closed as "not planned".

MCUdude, 2022-06-06:
> @MCUdude or dl8dtl Maybe this can be closed as "not planned".

Yes, this is not really a priority at the moment. PRs are always welcome though!

closed 2022-06-06 by MCUdude