avrdudes/avrdude #258: [bug #36713] config assumes obsolete libusb API

avrdude-admin, 2021-12-10:
Jeff Inman <avrdude.sieve@neverbox.com>
Sat 23 Jun 2012 10:02:55 PM UTC

If you build libusb 1.0.9, and then using that to build avrdude, you might have a config something like this (with tcsh):

(setenv CPPFLAGS -I.../libusb-1.0.9/installed/include/libusb-1.0; setenv LDFLAGS  -L.../libusb-1.0.9/installed/lib; setenv LIBS -lusb-1.0; ./configure --prefix=`pwd`/installed )

The last line in the build process would look something like this:

gcc -Wall -Wno-pointer-sign -g -O2  -L.../libusb-1.0.9/installed/lib -o avrdude avrdude-main.o avrdude-term.o ./libavrdude.a  -lusb-1.0   -lm -lreadline -lncurses -ltermcap -lusb-1.0

clearly, we're linking with libusb, but running avrdude with a usb programmer then spits out this message:

avrdude: error: no usb support. Please compile again with libusb installed.
avrdude: programmer operation not supported

avrdude done.  Thank you.

Turns out the problem is this part of the configure file [line 4499]:

LIBS="-lusb  $LIBS"
cat confdefs.h - <<_ACEOF >conftest.$ac_ext
/* end confdefs.h.  */

/* Override any GCC internal prototype to avoid an error.
Use char because int might match the return type of a GCC
builtin and then its argument prototype would still apply.  */
#ifdef __cplusplus
extern "C"
#endif
char usb_get_string_simple ();
int
main ()
{
return usb_get_string_simple ();
;
return 0;
}
_ACEOF
if ac_fn_c_try_link "$LINENO"; then :
ac_cv_lib_usb_usb_get_string_simple=yes
else
ac_cv_lib_usb_usb_get_string_simple=no
fi

The problem is that get_string_simple() is obsolete libusb-1.0.([http://libusb.org/wiki/APIs](https://savannah.nongnu.orghttp://libusb.org/wiki/APIs) says the 0.1 API is "deprecated and unmaintained since many years")  There is apparently a libusb-compat-0.1 version that provides 0.1 API via the 1.0 substrate.

You probably have lots of other things to worry about, before updating your USB interaction.  But I find lots of evidence on the web of folks running into this and being frustrated.  I suppose a temporary improvement would be to produce a better warning, maybe something like this:

avrdude: error: no usb support via libusb-0.1. Please compile again with libusb-0.1 installed.
avrdude: programmer operation not supported

avrdude done.  Thank you.

Thanks for the tool, though.

Jeff

This issue was migrated from https://savannah.nongnu.org/bugs/?36713

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 18 Sep 2013 07:27:34 AM UTC

Well, not all supported platforms do offer a libusb-1.0 API.
(The last one I've heard of was/is OpenSolaris.)

All known libusb-1.0 implementations also ship with a 0.1
API compatibility wrapper.  As AVRDUDE doesn't need any of
the newer 1.0 features, it is not in a real need to rewrite
everything for the 1.0 API.

Meanwhile, parts of AVRDUDE actually use the 1.0 API (those
parts will be disabled on systems that don't provide it), and
the configuration status is clearly printed at the end:

```
Configuration summary:
----------------------
DO HAVE    libelf
DO HAVE    libusb
DO HAVE    libusb_1_0
DON'T HAVE libftdi1
DON'T HAVE libftdi
DON'T HAVE libhid
DO HAVE    pthread
DISABLED   doc
ENABLED    parport
DISABLED   linuxgpio

```
Why can't you use the 0.1 API wrapper in your case?

mariusgreuel, 2022-01-12:
Use the current CMake project instead.

closed 2022-01-12 by mariusgreuel