avrdudes/avrdude #266: [bug #37697] avrdude cannot manage non-standard speeds on Mac

avrdude-admin, 2021-12-10:
Bairesport <marquevichi@gmail.com>
Fri 09 Nov 2012 05:37:04 AM UTC

Hi guys!

Working with an Arduino board (ATmel1284P) whose bootloader communicates at 250000bps (yeah, 250 tousand) I have found the avrdude running in OSX cannot handle non-standard speeds.

After a little research I modified the ser_posix.c, recompile it, and now it works great with any rare speed (all the credits for the libosmocore team).

If you have some time take a look to the attached ser_posix.c

Thanks for your amazing work!

[file #26891: ser_posix.c](https://savannah.nongnu.org/bugs/download.php?file_id=26891)
[file #26898: ser_posix.diff](https://savannah.nongnu.org/bugs/download.php?file_id=26898)

This issue was migrated from https://savannah.nongnu.org/bugs/?37697

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 09 Nov 2012 05:50:05 AM UTC

Could you resend it as the output of a diff -u command between
the original and your modified version?

avrdude-admin, 2021-12-10:
Javier Marquevichi <bairesport>
Fri 09 Nov 2012 01:45:23 PM UTC

Modification NOT checked under Linux. Please test it.
Thanks.

(**[file #26898](https://savannah.nongnu.org/support/download.php?file_id=26898)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 10 Sep 2013 09:27:26 PM UTC

I just had a look under Darwin 11.4.2.  The manual page
TCSETATTR(3) says:

```
The unsigned integer
speed_t is typedef'd in the include file <termios.h>.  The value of the integer corresponds
directly to the baud rate being represented; however, the following symbolic values are
defined:

```
(The various Bxxx constants then directly map to the value
xxx, respectively.)

To me, this looks as if the current codebase should work as
expected.  Is it possible that recent OSX releases changed
their behaviour here?

The problem with the patch is that it **only** accepts OSX and
Linux (no *BSD, no Solaris), and that redefining an ioctl
within application code is basically a no-go.  Nobody would
guarantee the ioctl is still encoded the same way in the next
OS release.

MCUdude, 2022-03-19:
This issue will be resolved in #898

commit 99f191a0 (2022-03-19): Merge pull request #898 from dl8dtl/macos-nonstandard-baudrates

Macos nonstandard baudrates

closes #266
closed 2022-03-19 by dl8dtl