avrdudes/avrdude #268: [bug #37768] Poll usbtiny 100 times at init time to handle low-clock devices

avrdude-admin, 2021-12-10:
keithp <None>
Sun 18 Nov 2012 07:21:16 PM UTC

I've got an ATtiny85 project that clocks the CPU down to 250kHz at boot time. That's too slow for programming. To fix this, I need to hold RESET low when first powering up the device.

Here's a patch which polls the programmer 100 times at init time instead of polling only twice. As reset is held low while polling, this allows me to start avrdude and then power up the device. As soon as the power switch is flipped, programming starts.

Devices which aren't broken like this won't see any changes.

[file #26926: 0001-Poll-usbtiny-100-times-at-init-time-instead-of-twice.patch](https://savannah.nongnu.org/bugs/download.php?file_id=26926)
[file #29099: t45.c](https://savannah.nongnu.org/bugs/download.php?file_id=29099)
[file #29118: working.png](https://savannah.nongnu.org/bugs/download.php?file_id=29118)
[file #29117: clockchange-stk500.png](https://savannah.nongnu.org/bugs/download.php?file_id=29117)
[file #29116: init-stk500.png](https://savannah.nongnu.org/bugs/download.php?file_id=29116)
[file #29115: failing.png](https://savannah.nongnu.org/bugs/download.php?file_id=29115)

This issue was migrated from https://savannah.nongnu.org/bugs/?37768

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 10 Sep 2013 09:14:17 PM UTC

I'm not completely reluctant to accept the patch, but in
trying to reproduce the problem, I failed so far.

First, I picked an ATmega16, and clocked it externally down
to 100 kHz.  Running AVRDUDE with the USBtinyISP and the
argument -B100 (resulting in a 10 kHz ISP clock) allowed me
to successfully connect to it.

Then, I picked an ATtiny45, and changed the low fuse to
0xe4, resulting in it being clocked from the 128 kHz watchdog
oscillator.  Again, using -B100 gave me a success in
connecting.

Finally, I changed the low fuse of the 'tiny45 to 0x64, so
the 1:8 prescaler is active, resulting in a total CPU clock
of 16 kHz.  Obviously, -B100 is still too fast now, but
-B1000 (1 kHz ISP clock) makes it work again.

Am I missing something?

avrdude-admin, 2021-12-10:
Keith Packard <keithp>
Tue 10 Sep 2013 09:39:08 PM UTC

Hrm. I'm not using the fuse to adjust the clock, rather I'm letting the chip come up at full speed and then clocking it down once the OS starts. I wonder if this affects how it responds to the reset?

I can try messing with the fuses and see if that makes things happy; that would at least let us know where the difference lies.

Of course I understand why you're reluctant to take a patch which doesn't seem to have any effect that you can see :-)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 10 Sep 2013 09:47:18 PM UTC

> I wonder if this affects how it responds to the reset?

It probably does, but just the other way around: upon reset,
no application code is running, so the CPU clock is just
the clock as configured by the fuses (whatever it might be,
including the effect of the CKDIV8 fuse).

Maybe it's some other effect, like the device waking up
from sleep?

I don't pretend to fully understand the existing USBtinyISP
code though.  As I read it, it asserts /RESET rather early,
and that should certainly get the target into its
fuse-configured speed sooner or later.

If you can come up with a (firmware) testcase that
reproduces the issue, that would be welcome.  I'd like to
understand what's going on before changing the code (in
particular since I didn't write that code myself).

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 11 Sep 2013 07:20:56 AM UTC

For another test, I tried the attached testcase.  It sets
the clock prescaler to 32 (= 250 kHz clock using default
fuses), makes timer 1 overflow each 256 ms to toggle a LED,
and puts the device on sleep otherwise.

I need to specify -B20 on the commandline (50 kHz ISP clock),
but otherwise, the USBtiny works fine with this constellation.

OTOH, since after a reset, the device is being clocked with
the default clock according to the fuses (i.e., 1 MHz), it
is somewhat surprising that -B20 is needed which in turn
slows down the entire ISP.  I'll try analyzing that a bit more.

(**[file #29099](https://savannah.nongnu.org/support/download.php?file_id=29099)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 11 Sep 2013 09:27:20 AM UTC

Btw., applying your patch doesn't help here either, it still
requires the -B20 (where it will work even without the patch).

I'm afraid I have to fire the logic analyzer onto this ...

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 12 Sep 2013 09:16:50 AM UTC

OK, as threatened, I used a logic analyzer to see what's
going on.

I attached the LA to the ISP signals (/RESET, MISO, MOSI, SCK),
and to CLKO.  The controller had the CKOUT fuse set.  The idea
was to monitor when the clock actually changes back from the
prescaled one to the one as defined by the fuses.

Much to my surprise, when using the USBtiny, CKOUT disappeared
as soon as /RESET was pulled low, file "failing.png".  Thus,
it could not be noticed when the clock actually changes.

I then used the STK500 programmer module.  Interestingly, CKOUT
keeps present here even after /RESET pulled low, see
"init-stk500.png".  As already suspected, the clock remains
at the scaled (lower) value even after the reset.  That explains
why a slow ISP speed is needed once the firmware reduced the
main clock speed.

The actual clock change happens the moment the "ISP init"
sequence has been successfully decoded, see
"clockchange-stk500.png".  That means, the lower ISP clock
has to be maintained for just the init sequence, but could
be made faster later on.  This is currently not supported by
AVRDUDE, and implementing it requires some larger-scale
changes.

Therefore, the recommendation that can be given by now for
users where the firmware is running at reduced clock speed
is:

. use a large enough -B value
. if that's getting to slow for overall ISP operation, just
use that value to issue a chip erase (-e), and then
proceed with a second AVRDUDE session at higher speed; use
-D to avoid the automatic erase (as the device has been
erased just before)

I'll add something like this to the documentation.

Btw., see picture "working.png": the SCK on the USBtiny varies
in a fairly large scale.  While the -B value given on the
commandline is correctly reflected by the closest edges in the
SCK signal (markers G1/G2 and x/o, both displaying half of
one -B20 period), the average SCK clock speed is more like a
fourth or a fifth of the max clock.  This results in a much
slower operation than would be possible.

The reason why your patch might improve the situation in some
cases is probably that by repeatedly pulling/releasing /RESET,
one might sometimes hit a time window where the internal
reset processing causes the clock to be re-evaluated again
based on the fuse setting.  However, this does not seem to
be reliable in any way (that's why the patch didn't work for
me).

Given all these explanations, is it OK to close the bug?

[image]
[image]
[image]
[image]

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 12 Sep 2013 11:34:21 AM UTC

A respective paragraph has been added to doc/avrdude.texi,
SVN r1218.

mcuee, 2022-06-05:
@dl8dtl If the documentation has been udpated, can this be closed?

mcuee, 2022-06-22:
Documentation has been in place as seen here.
https://avrdudes.github.io/avrdude/7.0/avrdude_39.html#Troubleshooting

_Problem: after flashing a firmware that reduces the target’s clock speed (e.g. through the CLKPR register), further ISP connection attempts fail.

Solution: Even though ISP starts with pulling /RESET low, the target continues to run at the internal clock speed as defined by the firmware running before. Therefore, the ISP clock speed must be reduced appropriately (to less than 1/4 of the internal clock speed) using the -B option before the ISP initialization sequence will succeed.

As that slows down the entire subsequent ISP session, it might make sense to just issue a chip erase using the slow ISP clock (option -e), and then start a new session at higher speed. Option -D might be used there, to prevent another unneeded erase cycle._

closed 2022-06-22 by mcuee