avrdudes/avrdude #274: [bug #37997] AVR109 not working

avrdude-admin, 2021-12-10:
Christian Pohl <nongnu.org@chpohl.de>
Sat 29 Dec 2012 04:24:43 PM UTC

Moin moin allerseits,

ich möchte auf einen Bug in der (in den aktuellen -- 29.12.2012 -- Ubuntu-Repos vorhandenen) Version 5.11.1 von avrdude hinweisen:

Auf dem 29C3 des Chaos Computer Clubs im CCH in Hamburg habe ich von blinkenarea.org eine kleine Schaltung "OrnaMatrix" ([http://wiki.blinkenarea.org/index.php/OrnaMatrix](https://savannah.nongnu.orghttp://wiki.blinkenarea.org/index.php/OrnaMatrix)) und einen kleinen ISP vom gleichen Entwickler erstanden. Nach anfänglichen Schwierigkeiten konnten wir sicher stellen, dass keine Hardwarefehler vorhanden sind.
Die avrdude Version 5.11.1 auf meinem LinuxMint13 war nicht in der Lage, den ATTiny2313 mit einem neuen HEX-File zu versorgen. Erst nachdem ich aus dem Downloadbereich die Version 5.2 des avrdude geladen, configured, make, make installed habe, konnte ich den Flashvorgang erfolgreich durchführen.
Ich bin mir nicht sicher, welche weiteren Informationen zur Fehlersuche benötigt werden, stehe also gerne für Fragen zur Verfügung.

Gruß
Christian Pohl

[file #27164: avrdude_5_2_ok.log](https://savannah.nongnu.org/bugs/download.php?file_id=27164)
[file #27163: avrdude_5_11_1_failure.log](https://savannah.nongnu.org/bugs/download.php?file_id=27163)
[file #29049: avrdude_5.2_success.log](https://savannah.nongnu.org/bugs/download.php?file_id=29049)
[file #29048: avrdude_svn_r1210_error.log](https://savannah.nongnu.org/bugs/download.php?file_id=29048)

This issue was migrated from https://savannah.nongnu.org/bugs/?37997

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sat 29 Dec 2012 05:39:44 PM UTC

Communication language here is commonly English, as this is
an internation effort.

Please submit more details:

. what kind of programmer is it (the -c option)
. the complete commandline
. the log of both versions run with -vvvv (four times option
"-v"); note that all trace data go to stderr, so you have
to redirect FD 2: "avrdude ... 2> logfile.txt"; please
provide these logfiles as an attachment here

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Sat 29 Dec 2012 08:28:07 PM UTC

Did you try the current svn version? Maybe the bug is already fixed.

avrdude-admin, 2021-12-10:
Christian Pohl <chrisphl>
Sat 29 Dec 2012 08:56:11 PM UTC

At first sorry for my german bug report.

I connected theflasher again and flashed using both versions of avrdude, the outputs of each are attached.

And no, I didn't try the latest svn version but I'll do and report the result here soon.

Thanks so far!

Christian

(**[file #27163](https://savannah.nongnu.org/support/download.php?file_id=27163)**, **[file #27164](https://savannah.nongnu.org/support/download.php?file_id=27164)**)

avrdude-admin, 2021-12-10:
Christian Pohl <chrisphl>
Sun 30 Dec 2012 12:49:37 PM UTC

Here's my command I used to make the flashing:

avrdude -p t2313 -P /dev/ttyUSB0 -c avr109 -e -U flash:w:ornamatrix.hex -b 115200

Using the last trunk from SVN repository does not even start anything: "avrdude: error at /usr/local/etc/avrdude.conf:297: programmer type must be written as "id_type"". So it seems it does not come around with the "old style" config file. Using the avrdude.cong from latest trunk it worked to me axactly the same way as avrdude version 5.11.1 -- it fails with the same output (as it seems to me).

avrdude-admin, 2021-12-10:
Christian Pohl <chrisphl>
Sun 30 Dec 2012 01:14:05 PM UTC

I noticed that the failing attempts with versions 5.11.xyz seem to flash much slower than the succeeded attempt with version 5.2 so my guess: Is this issue related to [https://savannah.nongnu.org/bugs/?37965](https://savannah.nongnu.orghttps://savannah.nongnu.org/bugs/?37965)?
I don't know how to use stty to configure serial port at the moment so can't evaluate.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 03 Sep 2013 12:20:33 PM UTC

See **[comment #2](https://savannah.nongnu.org#comment2)**: please get us a full -vvvv log of both
versions.

Yes, the avrdude.conf format has changed in the current version
(and that's the major reason why the next version will be
called 6.0).  Remember that you can always specify an
avrdude.conf file that matches your binary using -C, e.g.

```
./avrdude -C avrdude.conf -c avr109 -P /dev/ttyUSB0 -p t2313 -vvvv -b 115200 -U ornamatrix.hex 2> logfile

```
Note that **[bug #38965](https://savannah.nongnu.org/bugs/?38965)** has been closed upon request by its
originator, since he couldn't see the bug anymore.

avrdude-admin, 2021-12-10:
Christian Pohl <chrisphl>
Sun 08 Sep 2013 04:06:30 PM UTC

Sorry for being late. Just posting the requested log files.

kind regards
Christian

(**[file #29048](https://savannah.nongnu.org/support/download.php?file_id=29048)**, **[file #29049](https://savannah.nongnu.org/support/download.php?file_id=29049)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 08 Sep 2013 09:07:35 PM UTC

One discrepancy immediately catches attention:

The "OK" log says:

```
Programmer Type : avr910

```
The "Failed" log says:

```
Programmer Type : butterfly

```
Given the large number of device codes that are supported by
the programming hardware, I really doubt this is actually a
bootloader.  (Bootloaders usually support exactly one device,
the device they are operating on.)

Did you try with both, -c butterfly (aka. -c avr109), as well
as -c avr910?  While both implementations share the same
roots, their behaviour is not exactly the same.

One of the consequences of this might be that in the working
log, we can see:

```
avrdude_5_2: Send: T [54] ^ [5e]

```
i.e. the device code 0x5e is chose, which is the correct one
for an ATtiny2313.  In the "failed" log, we see:

```
avrdude: Send: T [54] . [13]
avrdude: Recv: . [0d]
avrdude: devcode selected: 0x13

```
i.e. it picks device code 0x13 which belongs to an AT90S1200.
This is done since the bootloader implementation just picks
the very first device code it can see, since many bootloaders
simply don't spend much attention on the device codes they
are claiming to support, so picking the first one is usually
the best choice.

However, for a standalone programmer, there are subtle
differences in the programming algorithm between AVR's
grandfather AT90S1200, and the ATtiny2313, which might explain
the failure.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 09 Sep 2013 03:33:12 PM UTC

Regarding my remark in **[comment #8](https://savannah.nongnu.org#comment8)**, here is the respective
part of the SVN log:

r718 | joerg_wunsch | 2007-01-11 21:20:25 +0100 (Do, 11. Jan 2007) | 10 Zeilen

- butterfly.c (butterfly_initialize): when sending the 'T'command (which is ignored by current AVR109 bootloaders),
send the first reply from the list of supported device
codes back rather than using avrdude.conf's idea about
an AVR910 device code.  Apparently, this solves disagreements
between different versions of at least the ATmega8 AVR910
device code.

Closes **[bug #18727](https://savannah.nongnu.org/bugs/?18727)**: Writing flash failed

Given the date of the change, the first AVRDUDE release
affected by it is 5.4.  This would match your observation
that 5.2 is working fine.

mcuee, 2022-06-05:
Not so sure if this is relevant.

mcuee, 2022-06-05:
Recently I had programmed the ATtiny2313 using avr910 (on the STK500) and it works. Unfortunately the STK500 board is now broken when I removed the ATtiny2313 (trying to program the ATmega8535) from the board (quite some traces broken due to the age of the board) so that I can not test it again.

mcuee, 2022-07-16:
Google translation of the initial report.
```
Hello everyone,

I would like to point out a bug in version 5.11.1 of avrdude (available in the current -- 12/29/2012 -- Ubuntu repos):

On the 29C3 of the Chaos Computer Club in the CCH in Hamburg I bought a small circuit "OrnaMatrix" (http://wiki.blinkenarea.org/index.php/OrnaMatrix) from blinkenarea.org and a small ISP from the same developer. After initial difficulties, we were able to ensure that there were no hardware errors.

The avrdude version 5.11.1 on my LinuxMint13 was not able to supply the ATTiny2313 with a new HEX file. Only after I loaded version 5.2 of avrdude from the download area, configured, make, make installed, was I able to carry out the flash process successfully.

I'm not sure what additional troubleshooting information is needed, so I'm happy to answer any questions.
```

mcuee, 2022-07-18:
@dl8dtl 
For the purpose of avrdude, `avr109, avr911 and butterfly` are the same, right?
AVR911 -- AVR open source programmer which supports AVR109 and AVR910
AVR109 -- self programming bootloader
AVR910 -- simple ISP programmer

https://github.com/avrdudes/avrdude/blob/2c0ccfed3a4745b5b50c2b2757e06b3f4b331b87/src/butterfly.c#L28
```
 * Actually, the butterfly uses a predecessor of the avr910 protocol
 * which is described in application notes avr109 (generic AVR
 * bootloader) and avr911 (opensource programmer).  This file now
 * fully handles the features present in avr109.  It should probably
 * be renamed to avr109, but we rather stick with the old name inside
 * the file.  We'll provide aliases for "avr109" and "avr911" in
 * avrdude.conf so users could call it by these name as well.
 */
```

mcuee, 2022-07-26:
I will close this issue as it is most likely an invalid report -- the reporter should have used `avr910` prorammer and not `avr109` bootloader.

closed 2022-07-26 by mcuee