avrdudes/avrdude #286: [bug #38831] [PATCH] Fix avrftdi onchip-buffer overrun

avrdude-admin, 2021-12-10:
Herm <None>
Fri 26 Apr 2013 01:03:20 AM UTC

The FT2232 only has a 384 byte fifo, but avrftdi tries to send arbitrarily large packets. Reading flash memory with page_size > 380 (example: mega16 has 512) fails.
The attached patch fixes this problem. It also makes sure the error code is checked.

[file #27942: fifo_overrun.patch](https://savannah.nongnu.org/bugs/download.php?file_id=27942)

This issue was migrated from https://savannah.nongnu.org/bugs/?38831

avrdude-admin, 2021-12-10:
Hannes Weisbach <hweisbach>
Sat 27 Apr 2013 10:00:05 PM UTC

SVN r1142 contains a patch similar to the one submitted by Hermann Kraus. Additionally it takes the chips buffer size into account, so that newer H-type chips do not experience a performance hit.
I tested this with {OS X 10.6.8, Ubuntu 12.04.2 and Win7 x64} x {FT2232C, FT4232H}.

avrdude-admin, 2021-12-10:
Tom <hrbn2>
Tue 30 Apr 2013 01:07:47 PM UTC

Look like it's working with FT2232D again (tested with Linux), but the support for the FT2232D is now disabled in code. In "avrftdi.c:747-751" between Svn-Version 1141 and 1142 the #if was removed and the FT2232D chip-type is 1 (TYPE_BM).

This chip is working for me with a pin_limit of 7 and package size of 384. Please re-enable it.

mcuee, 2022-05-27:
I believe the issue has been fixed in the current avrdude head (correct me if I am wrong -- not good at reading codes)
Ref: https://github.com/avrdudes/avrdude/blob/main/src/avrftdi.c

Take note FT2232D is compatible with FT2232C and FT2232L.
https://ftdichip.com/products/ft2232d/

MCUdude, 2022-05-27:
This is something @mariusgreuel or @dl8dtl has to look into. I have very little experience with Avrude + FT2232.

mcuee, 2022-06-02:
Ping @mariusgreuel and @dl8dtl

mcuee, 2022-06-08:
>The FT2232 only has a 384 byte fifo, but avrftdi tries to send arbitrarily large packets. Reading flash memory with page_size > 380 (example: mega16 has 512) fails.

BTW, from avrdude.conf, flash page size of ATmega16 is 128. I am not so sure about the first report.

The only chip with flash page size is the AVR32.
```

#------------------------------------------------------------
# AVR32UC3A0512
#------------------------------------------------------------

part
    id		= "uc3a0512";
    desc	= "AT32UC3A0512";
    signature	= 0xED 0xC0 0x3F;
    has_jtag	= yes;
    is_avr32	= yes;

    memory "flash"
        paged           = yes;
        page_size       = 512;           # bytes
        readsize        = 512;           # bytes
        num_pages       = 1024;          # could be set dynamicly
        size            = 0x00080000;    # could be set dynamicly
        offset          = 0x80000000;
    ;
;
```

mcuee, 2022-06-08:
@dl8dtl and @mariusgreuel 
Based on my tests in #474, this issue #286 is no longer present. I think you can close this.

closed 2022-06-08 by MCUdude