avrdudes/avrdude #288: [bug #38952] AVR109 (butterfly) auto-increment functionnality is ignored

avrdude-admin, 2021-12-10:
Magister <None>
Thu 09 May 2013 03:39:27 PM UTC

the butterfly_initialize() function check for the auto-increment feature of the bootloader using the 'a' command but ignore it by after.

Instead of sending multiple "A nn nn" command before each data packet, it could just send a "A 00 00" at the beginning and a flow of data packets by after, as the bootloader increment itself the address.

This would be way more efficient.

[file #28684: butterfly.c](https://savannah.nongnu.org/bugs/download.php?file_id=28684)

This issue was migrated from https://savannah.nongnu.org/bugs/?38952

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 03 Sep 2013 09:18:38 PM UTC

Since normal -U read/write operations are done page-wise,
the 'A' command is only issued once per page.  Keeping
track of each page's address between invocations to
butterfly_paged_write()/butterfly_paged_load() would cause
additional effort (including maintaining state within
library code).  I think the performance gain is not that
magnificent (less than 10 %) to warrant this change.

Opinions?

mcuee, 2022-06-07:
@dl8dtl Since the OP has not come back and it is has not been implemted for so many years and the benefits are not much, probably this can be closed as "Not Planned".

MCUdude, 2022-06-08:
I agree. PRs are always welcome, but there are no plans to add this feature at the moment.

closed 2022-06-08 by MCUdude