avrdudes/avrdude #289: [bug #39033] Support for ATxmegaXXXA4U devices

avrdude-admin, 2021-12-10:
Brian Mayton <None>
Tue 21 May 2013 06:30:25 PM UTC

avrdude appears to be able to communicate with an ATxmega128A4U in PDI mode, but an attempt to program flash memory fails at verification; the memory appears to read back as 0xFF.

It looks like avrdude can read the flash correctly; if I program the chip using Atmel Studio, a quick glance at 'dump flash' in avrdude's terminal mode looks like an intact interrupt table.  A chip erase from avrdude also appears to work; the flash reads back as 0xff after executing avrdude -e.

I captured the attached traces from Atmel Studio successfully programming (and then reading back) the flash memory.  The hex file used is the appropriate xmega DFU bootloader image from AVR1916.  The files were captured with Atmel Studio 6 running in VMware and wireshark on Linux capturing with the usbmon facility.  I would be happy to capture additional operations if it would be useful.

[file #28129: atxmega128a4u_dumps.tar.bz2](https://savannah.nongnu.org/bugs/download.php?file_id=28129)

This issue was migrated from https://savannah.nongnu.org/bugs/?39033

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 13 Sep 2013 02:16:15 PM UTC

After someone just got me an Xmega-A3BU Xplained board, I
gave it a try, and had no troubles (beyond the rather
complicated board setup) to talk PDI to it, using a JTAGICE3.

mcuee, 2022-06-08:
@dl8dtl There are many xmega related issues, not so sure this one (as well as others) is still relevant or not.

I have tried to link the xmega related issues here:
* https://github.com/avrdudes/avrdude/issues/435#issuecomment-1140370771

mcuee, 2022-08-17:
I just got an ATxmega32A4U breakout (from MCUZone) and there is no issue to program the flash using AVRISP mkii clone. And the device works fine after the programming (it has two buttons and two LEDs).

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -c avrispmkii -p x32a4u 
-U flash:r:atxmega32a4udemo_mcuzone.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: reading flash memory ...

Reading | ################################################## | 100% 0.67s

avrdude.exe: writing output file atxmega32a4udemo_mcuzone.hex

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -c avrispmkii -p x32a4u -e

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: erasing chip

avrdude.exe done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude.exe -c avrispmkii -p x32a4u
 -U flash:w:atxmega32a4udemo_mcuzone.hex:i

avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e9541 (probably x32a4u)
avrdude.exe: NOTE: Programmer supports page erase for Xmega devices.
             Each page will be erased before programming it, but no chip erase is performed.
             To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude.exe: reading input file atxmega32a4udemo_mcuzone.hex for flash
avrdude.exe: writing 36786 bytes flash ...

Writing | ################################################## | 100% 3.00s

avrdude.exe: 36786 bytes of flash written
avrdude.exe: verifying flash memory against atxmega32a4udemo_mcuzone.hex

Reading | ################################################## | 100% 0.67s

avrdude.exe: 36786 bytes of flash verified

avrdude.exe done.  Thank you.

```

mcuee, 2022-08-17:
I will close this issue as invalid.

closed 2022-08-17 by mcuee