avrdudes/avrdude #293: [bug #39690] Erase-rewrite cycle counter non-functional for ATxmega parts

avrdude-admin, 2021-12-10:
David Glaze <None>
Sun 04 Aug 2013 04:33:21 PM UTC

There are several issues I have found with using avrdude's erase-rewrite cycle counter (-y) functionality with ATxmega parts.  (I have only confirmed this with an ATxmega128A3U, but I strongly suspect the issue would be the same for other ATxmega models.)

The ATxmega parts appear to support both an atomic write operation and a "split" write operation for the EEPROM, where an erase occurs automatically only for the atomic write.  The EEPROM byte writing operations issued by Avrdude for the cycle counter appear to trigger a split write operation on the device, meaning that an explicit byte erase must be performed beforehand.  This is not done by Avrdude, as it only supports a page erase.

This behavior can be observed by doing a complete erase of the EEPROM, and then manually writing a value to the cycle counter address (-Y).  Any value chosen will be written successfully.  Any subsequent values written will only switch bits from 1 to 0.  Here is an attempt to write 15 (0x0F) and then write 240 (0xF0):

```
$ avrdude -c dragon_pdi -P usb -p x128a3 -v -Y 15

avrdude: Version 6.0rc1, compiled on Aug  4 2013 at 09:53:33
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/home/avarice/dave/local/etc/avrdude.conf"
User configuration file is "/home/avarice/dave/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : dragon_pdi
avrdude: usbdev_open(): Found AVRDRAGON, serno: 00A200034199
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
boot-loader FW version:        255
firmware version:              7.34
hardware version:              1
S_MCU:
boot-loader FW version:        255
firmware version:              7.34
hardware version:              7
Serial number:                   00:a2:00:03:41:99
Device ID:                       AVRDRAGON
AVR Part                      : ATxmega128A3
Chip Erase delay              : 0 us
PAGEL                         : P00
BS2                           : P00
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 0
StabDelay                     : 0
CmdexeDelay                   : 0
SyncLoops                     : 0
ByteDelay                     : 0
PollIndex                     : 0
PollValue                     : 0x00
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
prodsig        0     0     0    0 no        512  512      0     0     0 0x00 0x00
usersig        0     0     0    0 no        512  512      0     0     0 0x00 0x00
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
eeprom         0     0     0    0 no       2048   32      0     0     0 0x00 0x00
application    0     0     0    0 no     131072  512      0     0     0 0x00 0x00
apptable       0     0     0    0 no       8192  512      0     0     0 0x00 0x00
boot           0     0     0    0 no       8192  512      0     0     0 0x00 0x00
flash          0     0     0    0 no     139264  512      0     0     0 0x00 0x00
fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00

Programmer Type : DRAGON_PDI
Description     : Atmel AVR Dragon in PDI mode
M_MCU hardware version: 1
M_MCU firmware version: 7.34
S_MCU hardware version: 7
S_MCU firmware version: 7.34
Serial number:          00:a2:00:03:41:99
Vtarget         : 3.3 V

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9742
avrdude: setting erase-rewrite cycle count to 15

avrdude done.  Thank you.

```
```
$ ~/code/avrdude-svn/trunk/avrdude/avrdude -c dragon_pdi -P usb -p x128a3 -Y 240

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.15s

avrdude: Device signature = 0x1e9742
avrdude: current erase-rewrite cycle count is 15
avrdude: setting erase-rewrite cycle count to 240

avrdude done.  Thank you.

```
The last row of data from a hex dump of the EEPROM after the first and second operations above is:

```
:2007E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0000000F06

```
```
:2007E000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF0000000015

```
Note that the last 4 bytes should be 0x000000F0 after the second operation, and not 0x00000000.

In my local copy of avrdude, I have crudely hacked in a byte erase prior to a byte write for the EEPROM as a proof of concept, and confirmed that it works as expected for the AVR Dragon programmer.

The JTAGICE3 programmer has other issues on top of this.  There is a buffer overrun (that I will submit in a separate ticket), and an attempt to hack in similar support yields bizarre behavior where after issuing a MTYPE_EEPROM (0x22) write command, an error comes back saying that the MTYPE_SPM (0xA0) write command failed.  It's probably best if I don't submit my attempted fixes, to let the experts craft the correct solution for this reverse-engineered communication protocol.

This issue was migrated from https://savannah.nongnu.org/bugs/?39690

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 03 Sep 2013 12:05:45 PM UTC

Well, I asked about a couple of years ago whether anyone is
still using the erase cycle counter, and got zero responses.

I tend to remove that feature due to lack of usefulness.

I agree that EEPROM byte writes on Xmega are currently broken,
but I don't have the time to fix them right now.  If you want
to work with me on this, please let's discuss that on the
avrdude-dev mailinglist.

I'll keep the bug report open as a reminder.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 05 Sep 2013 09:20:45 PM UTC

I think I got the single-byte EEPROM updates working now
(SVN r1208).  Only tested for JTAG by now, hopefully will
also work on PDI.

Please give the SVN version a try if you can.

(I removed the erase cycle counter meanwhile, but single-byte
updates in terminal mode can be used to test this.)

MCUdude, 2022-03-25:
Erase counter is has not been supported since Avrdude 6.0.

closed 2022-03-25 by MCUdude