avrdudes/avrdude #303: [bug #40108] [PATCH] Fix avrftdi verification error after flash write

avrdude-admin, 2021-12-10:
Nemui <None>
Thu 26 Sep 2013 03:22:18 AM UTC

When write flash below sector size,usually polling byte cannot found.
In this case,verify flash after flash write always fail.

Tested with ATmega1284p(has 256byte sector size).
202byte(<256byte) binary fail,but over 256byte binaries was Ok.

Attached patch fixes this problem to re-initialize avrftdi when polling byte cannot be found.

[file #29220: success_patched_dude.log](https://savannah.nongnu.org/bugs/download.php?file_id=29220)
[file #29219: failure_orig_dude.log](https://savannah.nongnu.org/bugs/download.php?file_id=29219)
[file #29218: avrdude_avrftdi_fix.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29218)
[file #29223: avrftdi_sync.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29223)
[file #29858: avrftdi_testing_report.7z](https://savannah.nongnu.org/bugs/download.php?file_id=29858)

This issue was migrated from https://savannah.nongnu.org/bugs/?40108

avrdude-admin, 2021-12-10:
Hannes Weisbach <hweisbach>
Thu 26 Sep 2013 10:28:56 AM UTC

Thank you for reporting this.

Can you supply a log with -vvvvv as parameters to avrdude as well as the binary you are programming?
I am not accepting this patch as-is because (more or less) randomly resetting programmers is usually not the solution to the problem. In fact, the code is commented with a TODO, which already states what has to be done: /* TODO sync write */.
This means the flush part of write_flush() has to be called.

I've attached a patch (against r1248) implementing this TODO. Could you please try it and report back if they work (apply with -p1)?

(**[file #29223](https://savannah.nongnu.org/support/download.php?file_id=29223)**)

avrdude-admin, 2021-12-10:
Nemui Trinomius <nemuisan>
Wed 11 Dec 2013 04:49:07 AM UTC

Dear hweisbach,
Sorry for my late reply.

I tested your patch and got work well.
Please see attached report for detail.
(**[file #29858](https://savannah.nongnu.org/support/download.php?file_id=29858)**)

mcuee, 2022-06-05:
Maybe this can fix #364 as well.
* #364

mcuee, 2022-06-08:
I have both read and write issues using FT2232H based JTAGkey-2 with ATmega2560. I will try the patch mentioned in this thread. https://savannah.nongnu.org/support/download.php?file_id=29223

mcuee, 2022-06-25:
Most likely this issue has been fixed.

mcuee, 2022-07-10:
I will close this as `invalid` for now. Please re-open when the issue occurs.

closed 2022-07-10 by mcuee

mcuee, 2022-08-20:
The two patches are posted here for reference.

[file #29223: avrftdi_sync.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29223)

```
diff -urN b/avrftdi.c a/avrftdi.c
--- avrftdi.c	2013-09-20 09:25:29 +0900
+++ avrftdi.c	2013-09-20 09:48:28 +0900
@@ -49,6 +49,15 @@
 #define MIN(a,b) ((a)<(b)?(a):(b))
 #endif
 
+/*
+ * Verify error occurs after write flash with cannot find a poll byte situation!
+ * This situation caused by writing below first sector size.
+ * (e.g. Write below 256byte binary to ATMEGA1284p(has 256b SectorSize).)
+ * To re-initializing ftdi interface,verify action works well.
+ */
+static int reset_if =0;
+
+
 #ifdef DO_NOT_BUILD_AVRFTDI
 
 static int avrftdi_noftdi_open (struct programmer_t *pgm, char * name)
@@ -1081,6 +1090,8 @@
 		/* TODO sync write */
 		/* sleep */
 		usleep((m->max_write_delay));
+		/* Rize reset_if flag */
+		reset_if = 1;
 	}
 
 	return len;
@@ -1142,6 +1153,13 @@
 		buf_dump(o_buf, sizeof(o_buf), "o_buf", 0, 32);
 	}
 
+	/* Detect reset_if flag */
+	if(reset_if == 1){
+		/* Re-Initialize */
+		avrftdi_initialize(pgm,p);
+		reset_if =0;
+	}
+
 	if (0 > avrftdi_transmit(pgm, MPSSE_DO_READ | MPSSE_DO_WRITE, o_buf, i_buf, len * 4))
 		return -1;
 
```

mcuee, 2022-08-20:
[file #29218: avrdude_avrftdi_fix.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29218)
```
diff -urN b/avrftdi.c a/avrftdi.c
--- avrftdi.c	2013-09-20 09:25:29 +0900
+++ avrftdi.c	2013-09-20 09:48:28 +0900
@@ -49,6 +49,15 @@
 #define MIN(a,b) ((a)<(b)?(a):(b))
 #endif
 
+/*
+ * Verify error occurs after write flash with cannot find a poll byte situation!
+ * This situation caused by writing below first sector size.
+ * (e.g. Write below 256byte binary to ATMEGA1284p(has 256b SectorSize).)
+ * To re-initializing ftdi interface,verify action works well.
+ */
+static int reset_if =0;
+
+
 #ifdef DO_NOT_BUILD_AVRFTDI
 
 static int avrftdi_noftdi_open (struct programmer_t *pgm, char * name)
@@ -1081,6 +1090,8 @@
 		/* TODO sync write */
 		/* sleep */
 		usleep((m->max_write_delay));
+		/* Rize reset_if flag */
+		reset_if = 1;
 	}
 
 	return len;
@@ -1142,6 +1153,13 @@
 		buf_dump(o_buf, sizeof(o_buf), "o_buf", 0, 32);
 	}
 
+	/* Detect reset_if flag */
+	if(reset_if == 1){
+		/* Re-Initialize */
+		avrftdi_initialize(pgm,p);
+		reset_if =0;
+	}
+
 	if (0 > avrftdi_transmit(pgm, MPSSE_DO_READ | MPSSE_DO_WRITE, o_buf, i_buf, len * 4))
 		return -1;
 ``

mcuee, 2022-08-20:
Error log.
[file #29219: failure_orig_dude.log](https://savannah.nongnu.org/bugs/download.php?file_id=29219)

```
C:/Devz/AVR/avrdude/avrdude -v -p atmega1284p -P usb -c jtagkey    -B6 -U flash:w:main.hex 

avrdude: Version 6.1-svn-20130917, compiled on Sep 20 2013 at 09:27:17
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Devz\AVR\avrdude\avrdude.conf"

         Using Port                    : usb
         Using Programmer              : jtagkey
         Setting bit clk period        : 6.0
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : avrftdi
         Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9705
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "main.hex"
avrdude: input file main.hex auto detected as Intel Hex
avrdude: writing flash (202 bytes):

Writing | W avrftdi_flash_write(1078): No suitable byte (!=0xff) for polling found.
W avrftdi_flash_write(1079): Trying to sleep instead, but programming errors may occur.
W avrftdi_flash_write(1080): Be sure to verify programmed memory (no -V option)
################################################## | 100% 0.02s

avrdude: 202 bytes of flash written
avrdude: verifying flash memory against main.hex:
avrdude: load data flash data from input file main.hex:
avrdude: input file main.hex auto detected as Intel Hex
avrdude: input file main.hex contains 202 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.08s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x0c
avrdude: verification error; content mismatch

avrdude done.  Thank you.

```

mcuee, 2022-08-20:
[file #29220: success_patched_dude.log](https://savannah.nongnu.org/bugs/download.php?file_id=29220)

Good run log after patching avrdude.

```
C:/Devz/AVR/avrdude/avrdude -v -p atmega1284p -P usb -c jtagkey    -B6 -U flash:w:main.hex 

avrdude: Version 6.1-svn-20130917, compiled on Sep 20 2013 at 10:02:57
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Devz\AVR\avrdude\avrdude.conf"

         Using Port                    : usb
         Using Programmer              : jtagkey
         Setting bit clk period        : 6.0
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : avrftdi
         Description     : Amontec JTAGKey, JTAGKey-Tiny and JTAGKey2

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9705
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "main.hex"
avrdude: input file main.hex auto detected as Intel Hex
avrdude: writing flash (202 bytes):

Writing | W avrftdi_flash_write(1087): No suitable byte (!=0xff) for polling found.
W avrftdi_flash_write(1088): Trying to sleep instead, but programming errors may occur.
W avrftdi_flash_write(1089): Be sure to verify programmed memory (no -V option)
################################################## | 100% 0.02s

avrdude: 202 bytes of flash written
avrdude: verifying flash memory against main.hex:
avrdude: load data flash data from input file main.hex:
avrdude: input file main.hex auto detected as Intel Hex
avrdude: input file main.hex contains 202 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.17s

avrdude: verifying ...
avrdude: 202 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-08-20:
Testing report:
[file #29858: avrftdi_testing_report.7z](https://savannah.nongnu.org/bugs/download.php?file_id=29858)

Summary:

```
AVRDUDE Testing Report
20131211 Nemui Trinomius

Report URL:
 http://savannah.nongnu.org/bugs/?40108
Bug Summary:
 Verify error occurs after write with below sector size binary (using avrftdi).
 

Testing:
 Compare output logs pached/no-patched situation.
 See also test/environment.txt for this testing environment.

 *case1-1:
   Using no-patched avrdude v6.1 with -v-v-v option
   Write 204byte Binary(testprogram build with -Os option,below atmega1284p's flash sectorsize(256byte))
    ->Verify fail.(see test/case1-1.txt)
 *case1-2:
   Using no-patched avrdude v6.1 with -v-v-v option
   Write 1076byte Binary(testprogram build with -O0 option,above atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case1-2.txt)
 *case1-3:
   Using no-patched avrdude v6.1 with -v-v-v-v-v option
   Write 204byte Binary(testprogram build with -Os option,below atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case1-3.txt)
 *case1-4:
   Using no-patched avrdude v6.1 with -v-v-v-v-v option
   Write 1076byte Binary(testprogram build with -O0 option,above atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case1-4.txt)

 *case2-1:
   Using patched(Hannes Weisbach's patch) avrdude v6.1 with -v-v-v option
   Write 204byte Binary(testprogram build with -Os option,below atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case2-1.txt)
 *case2-2:
   Using patched(Hannes Weisbach's patch) avrdude v6.1 with -v-v-v option
   Write 1076byte Binary(testprogram build with -O0 option,above atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case2-2.txt)
 *case2-3:
   Using patched(Hannes Weisbach's patch) avrdude v6.1 with -v-v-v-v-v option
   Write 204byte Binary(testprogram build with -Os option,below atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case2-3.txt)
 *case2-4:
   Using patched(Hannes Weisbach's patch) avrdude v6.1 with -v-v-v-v-v option
   Write 1076byte Binary(testprogram build with -O0 option,above atmega1284p's flash sectorsize(256byte))
    ->Verify Succeed.(see test/case2-4.txt)

In "case1-3" must duplicate verify error,but verify succeeded....
Adding -v-v-v-v-v option cause syncwrite action due to buffer dump?

Conclusion:
 Avrftdi sync-write action seems work fine by Hannes Weisbach's patch.
```

mcuee, 2022-08-20:
New discussion in #1073. The patch mentioned in this thread is too old and no longer valid.