avrdudes/avrdude #304: [bug #40112] 6.0.1 - 911 flash upload fails

avrdude-admin, 2021-12-10:
Natan Keddem <nkeddem+avrdude@gmail.com>
Thu 26 Sep 2013 02:04:07 PM UTC

I am having some issues with the new windows version and programing an xmega128a1u with a 911 compatible bootloader.  I have attempted to use VirtualBox and Ubuntu 12.04.3 to build a and run avrdude 601 as well with identical results.  I have attached logs from both 5.11-**[Patch #7610](https://savannah.nongnu.org/patch/?7610)** and the "official" 6.0.1 build.  Both are windows versions.  5.11 works correctly and 6.0.1 fails.  Note that I am using the 128a1 not the 128a1u designator because 5.11 has no official support for the 128a1u and I want to keep everything as normalized as possible.

The command line used for both logs was the following:
avrdude -vvvv -V -c avr911 -p x128a1 -P com7 -b 19200 -U flash:w:b1.hex 2> <logfilenamehere>

It appears that after the signature is obtained there is no further actions taken by avrdude.  The program just leaves program mode and tells the bootloader to reset.  See the below table of transactions I took with a soft serial spy tool:

WRITE--1b-------"."
WRITE--53-------"S"
READ---42 74 4c 64 72-"BtLdr"
WRITE--56-------"V"
READ---31 30----"10"
WRITE--76-------"v"
READ---3f-------"?"
WRITE--70-------"p"
READ---53-------"S"
WRITE--61-------"a"
READ---59-------"Y"
WRITE--62-------"b"
READ---59-------"Y"
READ---02-------"."
READ---00-------"."
WRITE--74-------"t"
READ---7b-------"{"
READ---00-------"."
WRITE--54 7b----"T{"
READ---0d-------"."
WRITE--50-------"P"
READ---0d-------"."
WRITE--73-------"s"   <-Signature check
READ---4c 97 1e-"L—." <-Signature reply
WRITE--4c-------"L"   <-Exiting Program mode..why?!
READ---0d-------"."
WRITE--45-------"E"
READ---0d-------"."

[file #29225: loggood5117610-01.txt](https://savannah.nongnu.org/bugs/download.php?file_id=29225)
[file #29224: logbad601-01.txt](https://savannah.nongnu.org/bugs/download.php?file_id=29224)

This issue was migrated from https://savannah.nongnu.org/bugs/?40112

mcuee, 2022-06-06:
It seems to me this is similar to #311.
* #311

mcuee, 2022-07-18:
It mentions AVR911 but actually it is using the bootloader which is AVR109.

mcuee, 2022-08-06:
It may also be related to #360 and #404 since AVR109/911 bootloader is the issue for xmega.
* #360 
* #404

mcuee, 2022-08-08:
As per the test in #404 by @MCUdude, AVR109 bootloader (xboot) works fine with current avrdude for ATxemag128A3. So I believe this issue should have been fixed.
* https://github.com/avrdudes/avrdude/issues/454#issuecomment-1207482404

mcuee, 2022-08-08:
So I will close this issue as invalid now.

closed 2022-08-08 by mcuee