avrdudes/avrdude #307: [bug #40145] 6.0.1 - STK500v2 fails on uploads to parts with pages greater than 256b

avrdude-admin, 2021-12-10:
Natan Keddem <None>
Mon 30 Sep 2013 01:50:11 AM UTC

The function stk500v2_paged_write(and possibly other paged write functions) need the following changes:
1 - Start utilizing the blocksize variable in the .conf file to determine the maximum size of data transmissions.

2 - Perform sanity checks on blocksize to ensure conformance to STK500v2 transfer protocol maximum body specifications.  AVR068 states that the body should not exceed 275b, use this number or the buffer size you choose, which ever is less.

3 - Per AVR068, during a paged write transaction utilizing CMD_PROGRAM_FLASH_ISP  the mode byte will alert the programmer when a page should be written.

AVR068 states:
For paged operation, the Write page bit decides if a Write Program Memory Page command should be issued after the data has been loaded into the page buffer. For devices with page size bigger than what can be transferred to STK500 in one command, several CMD_PROGRAM_FLASH_ISP commands must be issued. In such a case, only the last command should have the Write Page
mode bit set.

[file #29263: doc2591.pdf](https://savannah.nongnu.org/bugs/download.php?file_id=29263)

This issue was migrated from https://savannah.nongnu.org/bugs/?40145

mcuee, 2022-06-03:
Ref: https://github.com/avrdudes/avrdude/blob/main/src/stk500v2.c
It seems to me the above issues may have already been addressed, at least partially. Maybe someone can take a look to see if this issue is still relevant or not.

1) Not so sure if avrdude.conf is used.

2) buffersize seems to be limited to 275 
https://github.com/avrdudes/avrdude/blob/main/src/stk500v2.c#L483
https://github.com/avrdudes/avrdude/blob/main/src/stk500v2.c#L2511

3) It seems this has been implemented. Correct me if I am wrong.
https://github.com/avrdudes/avrdude/blob/main/src/stk500v2.c#L2284

mcuee, 2022-06-20:
@dl8dtl 
When you got time, please take a look.

mcuee, 2022-06-25:
Most likely this has been fixed and no longer valid.

closed 2022-07-10 by mcuee