avrdudes/avrdude #310: [bug #40493] Can't program boot section of xmega256a3

avrdude-admin, 2021-12-10:
avrdude <leo.gartz@corenetix.com>
Wed 06 Nov 2013 04:14:29 PM UTC

I'm using avrdude 6.01 and JTAGICEmkII and try to programm the boot section
of the xmega256a3 with the following command:

avrdude -c jtag2pdi -P usb -p x256a3 -e -U flash:w:atxmega.hex:i

Output of avrdude:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9842
avrdude: erasing chip
avrdude: reading input file "atxmega.hex"
avrdude: writing flash (262232 bytes):

Writing | ################################################## | 100% 0.05s

avrdude: 262232 bytes of flash written
avrdude: verifying flash memory against atxmega.hex:
avrdude: load data flash data from input file atxmega.hex:
avrdude: input file atxmega.hex contains 262232 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.03s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x40000
0xff != 0xcf
avrdude: verification error; content mismatch

avrdude done.  Thank you.

When i use the same hex file and program my device with AtmelStudio 6.1
everything is ok. This means my JTAGICEmkII works correct.

[file #29547: atxmega.hex](https://savannah.nongnu.org/bugs/download.php?file_id=29547)

This issue was migrated from https://savannah.nongnu.org/bugs/?40493

mcuee, 2022-06-06:
This is probably related to #365
* https://github.com/avrdudes/avrdude/issues/365

mcuee, 2022-07-12:
This might be related to #337 though.
* #337

MCUdude, 2022-09-23:
I don't own a JTAGICEmkII, but the AVR Dragon is very similar. Here I'm flashing the stock DFU bootloader to an ATxmega256A3BU Xplained board using a dragon, without any issues. I think we can close this one.

```
$ ./avrdude -cdragon_jtag -p atxmega256a3bu -Uflash:w:/Users/hans/Downloads/AVR1916/XMEGA_bootloaders_v104/binaries/atxmega256a3u_104.hex -e

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: erasing chip
avrdude: reading input file /Users/hans/Downloads/AVR1916/XMEGA_bootloaders_v104/binaries/atxmega256a3u_104.hex for flash
         with 5612 bytes in 2 sections within [0x40000, 0x415f5]
         using 11 pages and 20 pad bytes
avrdude: writing 5612 bytes flash ...

Writing | ################################################## | 100% 0.00s

avrdude: 5612 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/AVR1916/XMEGA_bootloaders_v104/binaries/atxmega256a3u_104.hex

Reading | ################################################## | 100% 0.00s

avrdude: 5612 bytes of flash verified

avrdude done.  Thank you.
```

closed 2022-09-23 by MCUdude