avrdudes/avrdude #311: [bug #40518] Using application with AVR109 on Xmega  does not work as expected

avrdude-admin, 2021-12-10:
davidgrm <None>
Fri 08 Nov 2013 12:07:04 PM UTC

avrdude: Version 6.0.1, compiled on Sep 18 2013 at 08:20:41

F:\AvrDude_6>avrdude -p x64a3u -c avr911 -P COM2 -b 115200 -e  -U application:w:Timing1.hex -vvvv 2> logfile

A serial monitor shows this:

Port opened by process "avrdude.exe" (PID: 2148)

Request: 2013/11/08 14:01:16.48264

1B 53                                             .S

Answer: 2013/11/08 14:01:17.98264 (+0.0000 seconds)

58 6D 65 67 61 42 6C                              XmegaBl

Request: 2013/11/08 14:01:17.48264 (+0.5000 seconds)

56                                                V

Answer: 2013/11/08 14:01:17.48264 (+0.0000 seconds)

31 31                                             11

Request: 2013/11/08 14:01:17.48264 (+0.0000 seconds)

76                                                v

Answer: 2013/11/08 14:01:17.48264 (+0.0000 seconds)

3F                                                ?

Request: 2013/11/08 14:01:17.48264 (+0.0000 seconds)

70                                                p

Answer: 2013/11/08 14:01:17.49864 (+0.0156 seconds)

53                                                S

Request: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

61                                                a

Answer: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

59                                                Y

Request: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

62                                                b

Answer: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

59 01 00                                          Y..

Request: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

74                                                t

Answer: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

FA 00                                             ú.

Request: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

54 FA                                             Tú

Answer: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

0D                                                .

Request: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

50                                                P

Answer: 2013/11/08 14:01:17.49864 (+0.0000 seconds)

0D                                                .

Request: 2013/11/08 14:01:18.01464 (+0.5156 seconds)

73                                                s

Answer: 2013/11/08 14:01:18.01464 (+0.0000 seconds)

42 96 1E                                          B.

Request: 2013/11/08 14:01:18.01464 (+0.0000 seconds)

65                                                e

Answer: 2013/11/08 14:01:19.09264 (+1.0781 seconds)

0D                                                .

Request: 2013/11/08 14:01:20.36164 (+0.7656 seconds)

4C                                                L

Answer: 2013/11/08 14:01:20.36164 (+0.0000 seconds)

0D                                                .

Request: 2013/11/08 14:01:20.36164 (+0.0000 seconds)

45                                                E

Answer: 2013/11/08 14:01:20.36164 (+0.0000 seconds)

0D                                                .

Port closed

Avrdude log shows

Connecting to programmer: .avrdude: Send: . [1b]
avrdude: Send: S [53]
avrdude: Recv: X [58]

avrdude: Recv: m [6d] e [65] g [67] a [61] B [42] l [6c]
avrdude: Send: V [56]
avrdude: Recv: 1 [31] 1 [31]
avrdude: Send: v [76]
avrdude: Recv: ? [3f]
avrdude: Send: p [70]
avrdude: Recv: S [53]
Found programmer: Id = "XmegaBl"; type = S
Software Version = 1.1; No Hardware Version given.
avrdude: Send: a [61]
avrdude: Recv: Y [59]
Programmer supports auto addr increment.
avrdude: Send: b [62]
avrdude: Recv: Y [59]
avrdude: Recv: . [01]
avrdude: Recv: . [00]
Programmer supports buffered memory access with buffersize=256 bytes.
avrdude: Send: t [74]

Programmer supports the following devices:
avrdude: Recv: . [fa]
Device code: 0xfa
avrdude: Recv: . [00]

avrdude: Send: T [54] . [fa]
avrdude: Recv: . [0d]
avrdude: devcode selected: 0xfffffffa
avrdude: Send: P [50]
avrdude: Recv: . [0d]
avrdude: AVR device initialized and ready to accept instructions

Reading | avrdude: Send: s [73]
avrdude: Recv: B [42] . [96] . [1e]
################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9642
avrdude: erasing chip
avrdude: Send: e [65]
avrdude: Recv: . [0d]
avrdude: reading input file "Timing1.hex"
avrdude: input file Timing1.hex auto detected as Intel Hex
avrdude: writing application (28226 bytes):

Writing |  ***failed;
***failed;

This issue was migrated from https://savannah.nongnu.org/bugs/?40518

mcuee, 2022-07-16:
The title is actually not correct, it is about AVR911, similar to #304, and not about AVR109.

mcuee, 2022-08-14:
As per the test in #404 by @MCUdude, AVR109 bootloader (xboot) works fine with current avrdude for ATxemag128A3. So I believe this issue may not be present any more. It could be also the bootloader issue as well. The user is recommended to use xvoot for xmega (and not plain AVR109 exampple codes from Microchip).
* https://github.com/avrdudes/avrdude/issues/454#issuecomment-1207482404

closed 2022-08-14 by mcuee