avrdudes/avrdude #313: [bug #40615] Doesn't work with JTAGICE3 firmware 3.18

avrdude-admin, 2021-12-10:
Dennis Lambe <None>
Mon 18 Nov 2013 01:09:42 AM UTC

I used Atmel Studio 6.1 to upgrade the firmware on my JTAGICE3 from 2.15 to 3.8. After doing so, AVRDUDE begins reporting:

avrdude: usbdev_open(): did not find any (matching) USB device "usb:"

I investigated, and it turns out that the USB descriptors for firmware 3.8 are completely different from 2.15.

The Product ID has changed from 0x2110 to 0x2140. There are now two interfaces: one HID interface named "JTAGICE3 CMSIS-DAP" (the same name as in the iProduct string) with an Interrupt endpoint IN/OUT pair; and one vendor-specific interface named "JTAGICE3 Data Gateway" with a Bulk endpoint IN/OUT pair.

I have attached the output of lsusb on the device.

This forum thread suggests that the Data Gateway interface isn't needed for AVR programming: [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1092644](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1092644)

I haven't been able to dig up any information about whether the protocol on the HID Interrupt endpoints has anything at all to do with the 2.x USB protocol.

Maybe there are some clues in this release notice: [http://www.atmel.no/webdoc/jtagice3/jtagice3.whats_new.html](https://savannah.nongnu.orghttp://www.atmel.no/webdoc/jtagice3/jtagice3.whats_new.html)

[file #29646: lsusb-jtag3-3.8.txt](https://savannah.nongnu.org/bugs/download.php?file_id=29646)

This issue was migrated from https://savannah.nongnu.org/bugs/?40615

avrdude-admin, 2021-12-10:
Matthijs Kooijman <matthijs>
Fri 02 May 2014 09:10:48 AM UTC

For anyone coming across this bug, also see this mailing list thread: [http://lists.nongnu.org/archive/html/avrdude-dev/2013-11/msg00008.html](https://savannah.nongnu.orghttp://lists.nongnu.org/archive/html/avrdude-dev/2013-11/msg00008.html)

It contains some (far from finished) patches, and talks about Atmel offering documentation under NDA for anyone wanting to work on this.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 02 May 2014 09:23:19 AM UTC

The non-NDA documentation is here:

[http://www.atmel.no/webdoc/protocoldocs/index.html](https://savannah.nongnu.orghttp://www.atmel.no/webdoc/protocoldocs/index.html)

Anyway, this is supposed to be already fixed in the current
release.

avrdude-admin, 2021-12-10:
Matthijs Kooijman <matthijs>
Fri 02 May 2014 09:39:13 AM UTC

Ah, indeed, I found r1278 which implements this. Cool!

MCUdude, 2022-03-19:
Doesn't appear to be a problem anymore.

closed 2022-03-19 by MCUdude