avrdudes/avrdude #314: [bug #40702] fix: enable AVRISPmkII to read prodsig

avrdude-admin, 2021-12-10:
Stephan Baerwolf <stephan@matrixstorm.com>
Mon 25 Nov 2013 03:13:43 PM UTC

[file #29709: 0003-fix-enable-AVRISPmkII-to-read-prodsig.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29709)

This issue was migrated from https://savannah.nongnu.org/bugs/?40702

MCUdude, 2022-03-19:
```patch
@@ -3984,7 +3984,8 @@ static int stk600_xprog_read_byte(PROGRAMMER * pgm, AVRPART * p, AVRMEM * mem,
         b[1] = XPRG_MEM_TYPE_FUSE;
     } else if (strncmp(mem->desc, "lock", strlen("lock")) == 0) {
         b[1] = XPRG_MEM_TYPE_LOCKBITS;
-    } else if (strcmp(mem->desc, "calibration") == 0) {
+    } else if (strcmp(mem->desc, "calibration") == 0 ||
+               strcmp(mem->desc, "prodsig") == 0) {
         b[1] = XPRG_MEM_TYPE_FACTORY_CALIBRATION;
     } else if (strcmp(mem->desc, "usersig") == 0) {
         b[1] = XPRG_MEM_TYPE_USERSIG;
```

I don't see how this is useful. None of the "classic" AVRs has a memory named "prodsig" in avrdude.conf.
@dl8dtl should we just close this?

dl8dtl, 2022-03-19:
Xmegas (and their more recent relatives AVR8X and AVR Dx) do have "prodsig".

MCUdude, 2022-05-30:
It's already possible to read `prodsig` using a Dragon and an AVRISPmkII. It looks like this patch has already been applied (line 3946).

https://github.com/avrdudes/avrdude/blob/feaa1c6a6b0a0cf234dbd9e86e5b0d13d745c09a/src/stk500v2.c#L3925-L3972

```
$ ./avrdude -p atxmega128a3u -c avrispmkii -Uprodsig:r:-:h

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9742 (probably x128a3u)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading prodsig memory:

Reading | ################################################## | 100% 0.00s

avrdude: writing output file "<stdout>"
0xc,0x40,0x75,0xa,0x40,0x9f,0xff,0x1,0x31,0x4a,0x30,0x37,0x33,0x30,0xff,0xff,0x5,0xff,0xe,0x0,0x7,0x0,0xff,0xff,0xff,0xff,0x7e,0xf8,0x24,0x40,0xff,0xff,0x0,0x0,0x0,0xff,0x0,0x0,0x0,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0x9f,0x9,0xff,0xff
```