avrdudes/avrdude #319: [bug #40800] Can't program efuse on ATmega164p

avrdude-admin, 2021-12-10:
Peter Gibson <None>
Wed 04 Dec 2013 01:06:22 AM UTC

Trying to write/read efuse on a ATmega164p results in:
"ef" memory type not defined for part "ATmega164P"

avrdude: Version 6.0.1, compiled on Nov 18 2013 at 11:16:38
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/opt/local/etc/avrdude.conf"

In avrdude v6.0.1 from macports, the system wide config file does not correctly define the efuse memory type for this part. In version 5.11 (included in Arduino application), the m164p has its own declaration, while in v6.0.1 the declaration is based on a parent part "m16" with some modifications. However the m16 doesn't support an efuse, and the new m164p declaration is missing the efuse definition (amongst other things).

I've attached a .avrduderc file with an amended m164p entry, however I'm not sure if this is completely correct. Perhaps someone with knowledge of the file format could check it?

In addition, the ocdrev value has changed from v5.11 to 3. Given the other errors, could someone confirm if this value is correct?

Thanks

[file #29781: avrduderc](https://savannah.nongnu.org/bugs/download.php?file_id=29781)

This issue was migrated from https://savannah.nongnu.org/bugs/?40800

avrdude-admin, 2021-12-10:
Denis Sergeevich <galilley>
Mon 12 Mar 2018 09:56:15 AM UTC

avrdude version 6.3
the bug is still here

MCUdude, 2022-01-07:
I doubt this is an issue anymore. I've programmed and sat fuses the ATmega164P many times with Avrdude 6.3.

Atmega164P is now based on the ATmega324P implementation in Avrdude.conf.in

Issue can probably be closed.

https://github.com/avrdudes/avrdude/blob/7d574ccf0bea5d4e80dee701b923398714c14d49/src/avrdude.conf.in#L4709-L4731

closed 2022-01-08 by dl8dtl