avrdudes/avrdude #321: [bug #40831] LUFA AVRISP-MKII fails with avrdude 6.0.1

avrdude-admin, 2021-12-10:
Larry Viesse <None>
Sat 07 Dec 2013 03:10:06 AM UTC

The problem was found in usbdev_open in file usb_libusb.c.

The if statement:

```
if (fd->usb.rep) == 0)
{

```
prevents Dean's LUFA masqueraded endpoint from being discovered.

See full details at: [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&t=138424&start=0&postdays=0&postorder=asc&highlight](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&t=138424&start=0&postdays=0&postorder=asc&highlight)=

specifically in this post: [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1119592#1119592](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1119592#1119592)

Thanks for providing us with this awesome software.

[file #32171: endpointdetect_pass1.patch](https://savannah.nongnu.org/bugs/download.php?file_id=32171)

This issue was migrated from https://savannah.nongnu.org/bugs/?40831

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 12 Mar 2014 09:17:40 PM UTC

It's still not quite clear to me why this happens.

The statement you quoted is only supposed to trigger if the
read endpoint number (usb.rep) has not been set by the driver
before (== 0).  An endpoint number of 0 is always reserved for
the control endpoint, and cannot be a data endpoint anyway.
Then, a (half-hearted) attempt is made inside usb_libusb.c to
pick the first data endpoint in the configuration that has the
"IN" direction bit (0x80) set, and this one is used for usb.rep.
(Arguably, this procedure still leaves a lot room for
improvements, yes.)

Can you please post a full USB descriptor from the LUFA
firmware?

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 12 Mar 2014 10:26:43 PM UTC

OK, I see it in your file now:

The LUFA firmware **requires** the EP guesswork heuristics to
work.

OK, this really asks for doing it right then.

avrdude-admin, 2021-12-10:
Dean Camera <abcminiuser>
Sat 13 Sep 2014 07:50:29 AM UTC

Looking over the source code, it appears the issue is that the USB endpoints are already pre-filled by the upper tool layer:

(In STK500v2.c:)
pgm->fd.usb.rep = USBDEV_BULK_EP_READ_MKII;
pgm->fd.usb.wep = USBDEV_BULK_EP_WRITE_MKII;

That means the read endpoint heuristics inside usb_libusb.c will not trigger for known Atmel tools, and the code will assume that the read endpoint is always equal to the tool defaults. This breaks for LUFA programmers in libUSB compatibility mode, as the read endpoint index is changed to 0x83 (but the descriptors are otherwise identical).

avrdude-admin, 2021-12-10:
Dean Camera <abcminiuser>
Sat 13 Sep 2014 07:53:41 AM UTC

Since there is already a read endpoint fallback in usb_libusb.c:

if (fd->usb.rep == 0)
{
avrdude_message(MSG_INFO, "%s: usbdev_open(): cannot find a read endpoint, using 0x%02x\n",
progname, USBDEV_BULK_EP_READ_MKII);
fd->usb.rep = USBDEV_BULK_EP_READ_MKII;
}

Should we either set the usb.rep value to 0 in the upper tool layer, or explicitly zero it out in the usb layer?

avrdude-admin, 2021-12-10:
Colin O  Flynn <c_oflynn>
Mon 22 Sep 2014 10:41:03 PM UTC

I've added a possible patch against SVN head. This gets rid of hard-coded endpoints & sizes for the stk500v2 and jtagice2 files. It's fairly simple, but I've tested with an Atmel JTAG-ICEMKII, LUFA AVR-ISPMKII in AVRStudio Mode, and LUFA AVR-ISPMKII in LibUSB mode.

I haven't modified the files for stk600 or jtagice3, which also have hard-coded endpoints. Wanted to get feedback on this approach first before going onward...

(**[file #32171](https://savannah.nongnu.org/support/download.php?file_id=32171)**)

avrdude-admin, 2021-12-10:
Michael Mauch <elmicha>
Sat 26 Dec 2015 08:01:24 PM UTC

Your [patch]([https://savannah.nongnu.org/support/download.php?file_id=32171](https://savannah.nongnu.orghttps://savannah.nongnu.org/support/download.php?file_id=32171)) also applies to RELEASE_6_2, and it makes avrdude-6.2 work with an Olimex AVR-ISP-MK2.

avrdude-admin, 2021-12-10:
Dylan Taft <dylanetaft>
Wed 27 Jan 2016 02:34:18 AM UTC

Confirmed, and the patch still works against 6.2.  Will it be accepted upstream?

avrdude-admin, 2021-12-10:
xi <xii>
Fri 24 Jun 2016 12:34:19 PM UTC

Hi, I don't understand why the patch is still not applied?

I have tested avrdude 6.3 and it still has the problem with Olimex AVR-ISP-MK2 programmer on Linux (Mageia 5).

Please apply this patch, none of the avrdude 6.x work with the Olimex programmer otherwise!
(note: tested with an official Atmel AvrISP MK2 and it also works)

avrdude-admin, 2021-12-10:
julian <comboy>
Tue 31 Jan 2017 02:04:40 PM UTC

patch works with 6.3
had to apply it for the LUFa programmer.
Strong Vote to include it!

avrdude-admin, 2021-12-10:
Hey Duds <heyduds1>
Sun 12 Mar 2017 12:11:19 AM UTC

Not sure why you are having a problem with LUFA + AVRISP MKII if this patch is not applied. I have not applied this patch, but I installed the latest LUFA firmware and my AVRISP MKII works with avrdude 6.2 as well as 6.3 . The only problem is it gives me avrdude: fileio: invalid operation=1
avrdude: write to file 'filename' failed
when I am setting an output file format as "immediate" (m)

avrdude-admin, 2021-12-10:
andreas <ante>
Thu 23 Mar 2017 10:01:53 AM UTC

Please apply this patch to the official build, none of the avrdude 6.x-versions work with the LUFA-based AVRISP-MKII Clone and most likely the many derrivatives too
 [http://www.fourwalledcubicle.com/AVRISP.php](https://savannah.nongnu.orghttp://www.fourwalledcubicle.com/AVRISP.php)

avrdude-admin, 2021-12-10:
julian <comboy>
Mon 27 Mar 2017 01:03:14 PM UTC

without the patch i get
"avrdude: stk500v2_recv_mk2: error in USB receive"
with avrdude: Version 6.2

the patch solves it.

avrdude-admin, 2021-12-10:
Yegor Yefremov <yegorich>
Fri 07 Apr 2017 02:43:23 PM UTC

What is the state of this issue? I can still apply this patch to current SVN trunk and I get both Atmel and Olimex devices working in Linux. I would really like to see this issue fixed in 6.4.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Fri 07 Apr 2017 02:49:40 PM UTC

The question is not as much as what the state of this bug, but
rather when there will be any resources to prepare the next
release.

As long there is nobody actively preparing the release, it
doesn't matter much whether the patch is already in the tree
or not.

avrdude-admin, 2021-12-10:
Yegor Yefremov <yegorich>
Mon 10 Apr 2017 07:21:24 AM UTC

Spare time is a costly thing ...

Having the patch in the tree means, more people using latest tree would test it (consciously or not).

As I have already pointed in my previous post I have both Atmel and Olimex devices. Primary avrdude usage is on Linux for ARM. The only Atmel controller I'm using is Atmega 168pa (-p m168p) So let me know, if I can help with testing.

Though it is not likely to happen any time soon, but converting the project to git would be awesome. Alone patch creation via "git format-patch", "binary" debugging via git bisect and of course git blame feature make git far more appropriate for collaborative projects. No speaking about "tig" utility.

mariusgreuel, 2022-01-12:
Programmer need to be updated to the latest LUFA firmware that corrects the endpoint addresses.

closed 2022-01-12 by mariusgreuel

Zuikkis, 2023-08-29:
> Programmer need to be updated to the latest LUFA firmware that corrects the endpoint addresses.

That's not really a good advice as there are no LUFA firmware binaries easily available. Most users will go to Olimex site which has only binaries from 2016 which still have this problem.

The easiest way to get working avrdude is to download 6.3 source from
http://download.savannah.gnu.org/releases/avrdude/

then install the patch from here:
https://savannah.nongnu.org/bugs/?40831

build, and then you have a working avrdude.

I don't understand all the fuss here, why isn't the patch included when it's obviously working?

Pasetone, 2024-08-20:
Would this solve the problem?
https://github.com/abcminiuser/lufa/tree/master/Projects/AVRISP-MKII

Zuikkis, 2024-08-20:
> Would this solve the problem? https://github.com/abcminiuser/lufa/tree/master/Projects/AVRISP-MKII

Honestly no idea. Until there is a binary release of the firmware I'm not going to bother, the patched avrdude 6.3 works great for me. I have heard the patch works with never releases as well.

The original bug report is from 2013, I wouldn't hold my breath waiting for avrdude folks to include the patch.. :)

Olimex site recommends patched avrdude as well:
https://www.olimex.com/Products/AVR/Programmers/AVR-ISP-MK2/open-source-hardware

MCUdude, 2024-08-20:
The vanilla [LUFA AVRISPmkII firmware](https://github.com/abcminiuser/lufa/tree/master/Projects/AVRISP-MKII) is built for the AT90USB1287, while the Olimex AVRISPmkII uses the AT90USB162. I'm not sure how @OLIMEX got their hardware to work with the LUFA firmware. they've only provided pre-compiled hex files and no source code. If they did provide the source code, we could look into how to update the LUFA part of the firmware. And IMO, it is Olimex's responsibility to keep their firmware up to date.

Pasetone, 2024-08-20:
OK, I see. I have been using a patched 6.3 for years now but have not tried the with 7.X.
Any idea why the patch is not included in the newer versions? There have been several updates.

stefanrueger, 2024-08-20:
> Any idea why the patch is not included in the newer versions? There have been several updates.

I have no doubt that this particular patch is useful for this particular hardware. One difficulty for maintainers often is to assess how the patch/PR affects *all the other* working hardware. Currently, avrdude knows 98 USB-based programmers, and changing a central piece of USB comms needs to be vvv carefully evaluated.

My suggestion is to create a PR and make a *convincing* argument that the PR not only does the *right thing* but also won't affect the host of USB programmers that *do* work.

MCUdude, 2024-08-20:
I have an Olimex AVRISPmkII at home. I'll see if I can build the LUFA AVRISPmkII firmware for the Olimex programmer, and see if that will sort out the issue. If it does, I'll recommend users to update their firmware (it can be done using `-c flip1`) instead of applying a patch to Avrdude.

Zuikkis, 2024-08-20:
> I have no doubt that this particular patch is useful for this particular hardware. One difficulty for maintainers often is to assess how the patch/PR affects _all the other_ working hardware. [...]

Quickly browsing through the patch, it changes error handling in usb_libusb.c

original:

if (fd->usb.rep == 0)

new:

if (fd->usb.rep < 0)

The old ==0 case is not handled at all after the patch. I wonder if this is a mistake, should latter be <=0 ?

But anyway it should be trivial to add BOTH these if's to the code. If  <0 happens only with Olimex then that code will not have any effect on other devices, because that case is not even handled by regular avrdude.

> My suggestion is to create a PR and make a _convincing_ argument that the PR not only does the _right thing_ but also won't affect the host of USB programmers that _do_ work.

I'm pretty sure it's the _right thing_ that has prevented the patch being accepted for the last 11 years.. It probably is much cleaner to fix it at firmware level.

Zuikkis, 2024-08-20:
Ah no, the <0 vs ==0 comes from the different initialization of the .rep field in  jtagmkII.c and stk500v2.c . But anyway for "working hardware" the "if (fd->usb.rep < 0)" check is never true.

stefanrueger, 2024-08-20:
@Zuikkis Thanks. You'd be vvv welcome to do a deep dive into the code, analyse the error handling, suggest a PR, make a crisp case as to *why* it's the right thing (it may be obvious to you but, alas not us), and ask @dl8dtl to review it. Jörg apparently co-authored the file some 20(!) years ago, so he's the most knowledgeable on the team about that.

Pasetone, 2024-08-20:
If this is a bug that was introduced in V6 and not present in V5 where other programmers were working fine, I can't see the problem with correcting it. Perhaps I am missing something here.......

MCUdude, 2024-08-20:
I manage to get the latest AVRISPmkII LUFA firmware compiled and working on the Olimex AVRISPmkII programmer.
I've attached all the source files. 

Can you try upgrading the firmware using the built-in flip bootloader like so? My Olimex AVRISPmkII programmer works perfectly fine with Avrdude v7.3 and the latest git main.

```
avrdude -cflip1 -pat90usb162 -Uflash:w:/path/to/AVRISP-MKII/AVRISP-MKII.hex:i -Ueeprom:w:/path/to/AVRISP-MKII/AVRISP-MKII.eep:i
```

If this works, then it's no need to "fix" Avrdude. After all, it isn't broken for the official AVRISPmkII

> If this is a bug that was introduced in V6 and not present in V5 where other programmers were working fine, I can't see the problem with correcting it. Perhaps I am missing something here.......

Unlike with older releases, we're now performing extensive testing with actual programmers and microcontrollers. If we do significant changed to a codebase that lots of different programmers rely on, we have to perform a bunch of different tests to make sure nothing broke.

[AVRISP-MKII-LUFA.zip](https://github.com/user-attachments/files/16679761/AVRISP-MKII.zip)

MCUdude, 2024-08-21:
I did test the firmware the Olimex AVRISPmkII was shipped with, and it did work perfectly fine with "vanilla" Avrdude 7.3 and git main. However, it is a good idea to update the firmware anyway.

I've been in contact with Olimex customer support regarding this issue, so hopefully they can start shipping programmers with updated firmware.

Zuikkis, 2024-08-21:
> I did test the firmware the Olimex AVRISPmkII was shipped with, and it did work perfectly fine with "vanilla" Avrdude 7.3 and git main. However, it is a good idea to update the firmware anyway. [...]

This is interesting. I have an old Olimex AVRISPmkII that most definitely does not work with unpatched avrdude (6.x or 7.x). It worked with avrdude 5.x. 

So apparently Olimex has fixed something in newer firmwares? But the latest firmware in their product page is from 2016 and that one does not work.

Good that you have contacted Olimex. Maybe they can put the updated firmware on the download page so everyone doesn't have to compile themself.

MCUdude, 2024-08-21:
> So apparently Olimex has fixed something in newer firmwares? But the latest firmware in their product page is from 2016 and that one does not work.

And they told me that they believed this had been fixed on the Avrdude side of things, since it worked with later versions of Avrdude with their hardware and the "old" firmware that they still ship.

But please try flashing the firmware I attached to an earlier post and see if this resolves your issue.

Pasetone, 2024-08-21:
This is what I get when I try to upgrade:

avrdude: Warning: Failed to read USB device string 1: error sending control message: Operation not permitted
avrdude: Warning: Failed to read USB device string 2: error sending control message: Operation not permitted
avrdude: Warning: Failed to read USB device string 3: error sending control message: Operation not permitted
avrdude: Warning: USB bDeviceClass = 255 (expected 254)
avrdude: Error: Failed to reset DFU state: error sending control message: Operation not permitted
avrdude: AVR device initialized and ready to accept instructions

Reading |                                                    | 0% 0.00savrdude: Error: DFU_DNLOAD failed: error sending control message: Operation not permitted
avrdude: Error: Failed to get DFU status: error sending control message: Operation not permitted
avrdude: error reading signature data for part "AT90USB162", rc=-1
avrdude: error reading signature data, rc=-1

avrdude done.  Thank you.

MCUdude, 2024-08-21:
Not sure why `-c flip1` doesn't work for you. You may have to run Avrdude as administrator. It definitely works for me, but I'm using a mac.
Anyways, try downloading Atmel Flip, as suggested over at the [Olimex product website](https://www.olimex.com/Products/AVR/Programmers/AVR-ISP-MK2/open-source-hardware).

stefanrueger, 2024-08-21:
> Operation not permitted

Guessing the new FW uses a different usbvid/usbpid and the OS doesn't know who has permission for this vid/pid, so not allowing access. That woul be an OS problem but see also A.1.4 Unix USB Permissions in [avrdude.pdf](https://github.com/avrdudes/avrdude/blob/main/avrdude.pdf)

Pasetone, 2024-08-22:
I got it flashed with dfu programmer.
It now works with 6.3 and 7.3.
Thanks for the build.

MCUdude, 2024-08-22:
Excellent! I'll let Olimex know about this.

Pasetone, 2024-08-22:
All the info on the Olimex site looks about ten years old. 
Anybody using Ubuntu 14.04 these days....... ? ;)

DanKoloff, 2024-08-23:
It was 8 years old. I've updated everything today. Units manufactured after 26.08.24 will come with latest LUFA, rest of units can be updated via Flip or dfu-programmer using the new hex from the updated archive at the web-page of AVR-ISP-MK2.

Thanks for pointing out about these problems.