avrdudes/avrdude #325: [bug #41292] Crash when uploading with JTAGICE3 [through USB 1.1]

avrdude-admin, 2021-12-10:
Guyzmo <savannah@m0g.net>
Tue 21 Jan 2014 05:20:30 PM UTC

<p>
Starting from yesterday, when I try to flash my custom Atmega2560 board using the JTAG ICE 3
on the ICSP/ISP header, I'm getting the following error, and the flasher is stuck with the
"flashing status" led on (core dump attached).
</p>
<p>
It was working perfectly fine before I had an issue with the cable from the flasher to the
ICSP header, that I changed. Since I changed that cable, it is working fine on other boards
(Atmega32u4 based).
</p>
<pre>
% /usr/local/bin/avrdude -D -C /usr/local/etc/avrdude.conf  -p atmega2560 -P usb -c jtag3isp -b 250000 -B0.3 -e -U flash:w:applet/Marlin.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.19s

avrdude: Device signature = 0x1e9801
avrdude: erasing chip
avrdude: reading input file "applet/Marlin.hex"
avrdude: input file applet/Marlin.hex auto detected as Intel Hex
avrdude: writing flash (114316 bytes):

Writing | ################################################## | 100% 11.17s

avrdude: 114316 bytes of flash written
avrdude: verifying flash memory against applet/Marlin.hex:
avrdude: load data flash data from input file applet/Marlin.hex:
avrdude: input file applet/Marlin.hex auto detected as Intel Hex
avrdude: input file applet/Marlin.hex contains 114316 bytes
avrdude: reading on-chip flash data:

Reading |                                                    | 0% 0.00s*** glibc detected *** /usr/local/bin/avrdude: free(): invalid next size (fast): 0x0000000000fe8080 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x76d76)[0x7f35a0dacd76]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7f35a0db1aac]
/usr/local/bin/avrdude[0x410957]
/usr/local/bin/avrdude[0x411ddf]
/usr/local/bin/avrdude[0x4122f9]
/usr/local/bin/avrdude[0x4130a5]
/usr/local/bin/avrdude[0x4090b7]
/usr/local/bin/avrdude[0x41be2e]
/usr/local/bin/avrdude[0x404bf2]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd)[0x7f35a0d54ead]
/usr/local/bin/avrdude[0x405659]
======= Memory map: ========
00400000-00461000 r-xp 00000000 08:02 24511709                           /usr/local/bin/avrdude
00661000-00662000 rw-p 00061000 08:02 24511709                           /usr/local/bin/avrdude
00662000-00668000 rw-p 00000000 00:00 0
00d3b000-01013000 rw-p 00000000 00:00 0                                  [heap]
7f359c000000-7f359c021000 rw-p 00000000 00:00 0
7f359c021000-7f35a0000000 ---p 00000000 00:00 0
7f35a0714000-7f35a0729000 r-xp 00000000 08:02 12451882                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f35a0729000-7f35a0929000 ---p 00015000 08:02 12451882                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f35a0929000-7f35a092a000 rw-p 00015000 08:02 12451882                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7f35a092a000-7f35a092c000 r-xp 00000000 08:02 12451866                   /lib/x86_64-linux-gnu/libdl-2.13.so
7f35a092c000-7f35a0b2c000 ---p 00002000 08:02 12451866                   /lib/x86_64-linux-gnu/libdl-2.13.so
7f35a0b2c000-7f35a0b2d000 r--p 00002000 08:02 12451866                   /lib/x86_64-linux-gnu/libdl-2.13.so
7f35a0b2d000-7f35a0b2e000 rw-p 00003000 08:02 12451866                   /lib/x86_64-linux-gnu/libdl-2.13.so
7f35a0b2e000-7f35a0b35000 r-xp 00000000 08:02 12451868                   /lib/x86_64-linux-gnu/librt-2.13.so
7f35a0b35000-7f35a0d34000 ---p 00007000 08:02 12451868                   /lib/x86_64-linux-gnu/librt-2.13.so
7f35a0d34000-7f35a0d35000 r--p 00006000 08:02 12451868                   /lib/x86_64-linux-gnu/librt-2.13.so
7f35a0d35000-7f35a0d36000 rw-p 00007000 08:02 12451868                   /lib/x86_64-linux-gnu/librt-2.13.so
7f35a0d36000-7f35a0eb6000 r-xp 00000000 08:02 12451873                   /lib/x86_64-linux-gnu/libc-2.13.so
7f35a0eb6000-7f35a10b6000 ---p 00180000 08:02 12451873                   /lib/x86_64-linux-gnu/libc-2.13.so
7f35a10b6000-7f35a10ba000 r--p 00180000 08:02 12451873                   /lib/x86_64-linux-gnu/libc-2.13.so
7f35a10ba000-7f35a10bb000 rw-p 00184000 08:02 12451873                   /lib/x86_64-linux-gnu/libc-2.13.so
[... 39 lines cut ...]
7f35a23fd000-7f35a23fe000 rw-p 00000000 00:00 0
7fff9925b000-7fff9927c000 rw-p 00000000 00:00 0                          [stack]
7fff9938c000-7fff9938d000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
[1]    25307 abort (core dumped)  /usr/local/bin/avrdude -D -C /usr/local/etc/avrdude.conf -p atmega2560 -P usb
% /usr/local/bin/avrdude -v

avrdude: Version 6.0.1, compiled on Dec 10 2013 at 16:49:18
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2009 Joerg Wunsch

System wide configuration file is "/usr/local/etc/avrdude.conf"
User configuration file is "/home/guyzmo/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

avrdude: no programmer has been specified on the command line or the config file
Specify a programmer using the -c option and try again

[1]    25734 exit 1     /usr/local/bin/avrdude -v
</pre>

[file #30718: avrdude](https://savannah.nongnu.org/bugs/download.php?file_id=30718)
[file #30719: avrdude.core.bz2](https://savannah.nongnu.org/bugs/download.php?file_id=30719)

This issue was migrated from https://savannah.nongnu.org/bugs/?41292

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Tue 21 Jan 2014 07:37:16 PM UTC

Too bad you don't have more traces.

Since none of us can reproduce it, I'll keep it open, but with
a low priority.

avrdude-admin, 2021-12-10:
Guyzmo <guyzmo>
Mon 24 Feb 2014 11:46:52 AM UTC

Ok, this time I ran avrdude in gdb, and showed the backtrace, and I also got a core file to share:

Starting program: /usr/local/bin/avrdude -D -C /usr/local/etc/avrdude.conf  -p atmega2560 -P usb -c jtag3isp -b 250000 -B0.1 -e -U flash:w:applet/Marlin.hex
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.19s

avrdude: Device signature = 0x1e9801
avrdude: erasing chip
avrdude: reading input file "applet/Marlin.hex"
avrdude: input file applet/Marlin.hex auto detected as Intel Hex
avrdude: writing flash (118782 bytes):

Writing | ################################################## | 100% 11.58s

avrdude: 118782 bytes of flash written
avrdude: verifying flash memory against applet/Marlin.hex:
avrdude: load data flash data from input file applet/Marlin.hex:
avrdude: input file applet/Marlin.hex auto detected as Intel Hex
avrdude: input file applet/Marlin.hex contains 118782 bytes
avrdude: reading on-chip flash data:

Reading |                                                    | 0% 0.00s*** glibc detected *** /usr/local/bin/avrdude: free(): invalid next size (fast): 0x0000000000915080 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x76d76)[0x7ffff69acd76]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6c)[0x7ffff69b1b1c]
/usr/local/bin/avrdude[0x410957]
/usr/local/bin/avrdude[0x411ddf]
/usr/local/bin/avrdude[0x4122f9]
/usr/local/bin/avrdude[0x4130a5]
/usr/local/bin/avrdude[0x4090b7]
/usr/local/bin/avrdude[0x41be2e]
/usr/local/bin/avrdude[0x404bf2]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xfd)[0x7ffff6954ead]
/usr/local/bin/avrdude[0x405659]
======= Memory map: ========
00400000-00461000 r-xp 00000000 08:02 24511709                           /usr/local/bin/avrdude
00661000-00662000 rw-p 00061000 08:02 24511709                           /usr/local/bin/avrdude
00662000-00941000 rw-p 00000000 00:00 0                                  [heap]
7ffff0000000-7ffff0021000 rw-p 00000000 00:00 0
7ffff0021000-7ffff4000000 ---p 00000000 00:00 0
7ffff6314000-7ffff6329000 r-xp 00000000 08:02 12451882                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7ffff6329000-7ffff6529000 ---p 00015000 08:02 12451882                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7ffff6529000-7ffff652a000 rw-p 00015000 08:02 12451882                   /lib/x86_64-linux-gnu/libgcc_s.so.1
7ffff652a000-7ffff652c000 r-xp 00000000 08:02 12456707                   /lib/x86_64-linux-gnu/libdl-2.13.so
7ffff652c000-7ffff672c000 ---p 00002000 08:02 12456707                   /lib/x86_64-linux-gnu/libdl-2.13.so
7ffff672c000-7ffff672d000 r--p 00002000 08:02 12456707                   /lib/x86_64-linux-gnu/libdl-2.13.so
7ffff672d000-7ffff672e000 rw-p 00003000 08:02 12456707                   /lib/x86_64-linux-gnu/libdl-2.13.so
7ffff672e000-7ffff6735000 r-xp 00000000 08:02 12456741                   /lib/x86_64-linux-gnu/librt-2.13.so
7ffff6735000-7ffff6934000 ---p 00007000 08:02 12456741                   /lib/x86_64-linux-gnu/librt-2.13.so
7ffff6934000-7ffff6935000 r--p 00006000 08:02 12456741                   /lib/x86_64-linux-gnu/librt-2.13.so
7ffff6935000-7ffff6936000 rw-p 00007000 08:02 12456741                   /lib/x86_64-linux-gnu/librt-2.13.so
7ffff6936000-7ffff6ab8000 r-xp 00000000 08:02 12455328                   /lib/x86_64-linux-gnu/libc-2.13.so
7ffff6ab8000-7ffff6cb7000 ---p 00182000 08:02 12455328                   /lib/x86_64-linux-gnu/libc-2.13.so
7ffff6cb7000-7ffff6cbb000 r--p 00181000 08:02 12455328                   /lib/x86_64-linux-gnu/libc-2.13.so
7ffff6cbb000-7ffff6cbc000 rw-p 00185000 08:02 12455328                   /lib/x86_64-linux-gnu/libc-2.13.so
7ffff6cbc000-7ffff6cc1000 rw-p 00000000 00:00 0
7ffff6cc1000-7ffff6ce6000 r-xp 00000000 08:02 12451892                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7ffff6ce6000-7ffff6ee5000 ---p 00025000 08:02 12451892                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7ffff6ee5000-7ffff6ee9000 r--p 00024000 08:02 12451892                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7ffff6ee9000-7ffff6eea000 rw-p 00028000 08:02 12451892                   /lib/x86_64-linux-gnu/libtinfo.so.5.9
7ffff6eea000-7ffff6f0b000 r-xp 00000000 08:02 12451888                   /lib/x86_64-linux-gnu/libncurses.so.5.9
7ffff6f0b000-7ffff710a000 ---p 00021000 08:02 12451888                   /lib/x86_64-linux-gnu/libncurses.so.5.9
7ffff710a000-7ffff710b000 r--p 00020000 08:02 12451888                   /lib/x86_64-linux-gnu/libncurses.so.5.9
[... 27 lines cut ...]
7ffff7ece000-7ffff7fd9000 rw-p 00000000 00:00 0
7ffff7ff9000-7ffff7ffb000 rw-p 00000000 00:00 0
7ffff7ffb000-7ffff7ffc000 r-xp 00000000 00:00 0                          [vdso]
7ffff7ffc000-7ffff7ffd000 r--p 0001f000 08:02 12456746                   /lib/x86_64-linux-gnu/ld-2.13.so
7ffff7ffd000-7ffff7ffe000 rw-p 00020000 08:02 12456746                   /lib/x86_64-linux-gnu/ld-2.13.so
7ffff7ffe000-7ffff7fff000 rw-p 00000000 00:00 0
7ffffffde000-7ffffffff000 rw-p 00000000 00:00 0                          [stack]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]

Program received signal SIGABRT, Aborted.
0x00007ffff6968475 in raise () from /lib/x86_64-linux-gnu/libc.so.6
(gdb) BT
[... cut ...]
#8  0x00000000004130a5 in stk500v2_paged_load (pgm=0x7072d0, p=<optimized out>, m=0x8309d0, page_size=256, addr=0, n_bytes=256) at stk500v2.c:2570
#9  0x00000000004090b7 in avr_read (pgm=pgm@entry=0x7072d0, p=p@entry=0x82eb80, memtype=0x668480 "flash", v=v@entry=0x90f0f0) at avr.c:412
#10 0x000000000041be2e in do_op (pgm=0x7072d0, p=p@entry=0x82eb80, upd=0x668450, flags=flags@entry=UF_NONE) at update.c:348
#11 0x0000000000404bf2 in main (argc=<optimized out>, argv=<optimized out>) at main.c:1217

avrdude-admin, 2021-12-10:
Guyzmo <guyzmo>
Mon 24 Feb 2014 11:49:08 AM UTC

the only way I finally found to solve the issue, is to restart my computer...

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 27 Feb 2014 02:06:25 PM UTC

> the only way I finally found to solve the issue, is to
> restart my computer...

This is kinda strange.

> and I also got a core file to share

Please share, together with the executable.  I assume the
source code is stock version 6.0.1?  If you have locally
modified it, please also submit the source code.

I am not able to reproduce this in any way myself, so debugging
your executable and core file is the only way to get a grip at
it.

avrdude-admin, 2021-12-10:
Guyzmo <guyzmo>
Thu 27 Feb 2014 02:16:26 PM UTC

> > the only way I finally found to solve the issue, is to
> > restart my computer...
> This is kinda strange.

Because of that, I really feel this is not necessarily a bug
of avrdude itself, but a bug upstream, somewhere in the USB
libraries or the kernel drivers. So if we could find where
does that come from, I could make a bug report where it needs.

> > and I also got a core file to share
> Please share, together with the executable. I assume the
> source code is stock version 6.0.1? If you have locally
> modified it, please also submit the source code.

avrdude: Version 6.0.1, compiled on Dec 10 2013 at 16:49:18
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2009 Joerg Wunsch

> I am not able to reproduce this in any way myself, so debugging
> your executable and core file is the only way to get a grip at

I actually thought I had uploaded the core file... I actually
somehow find the savannah bug report interface slightly unergonomic
to use.. Anyway, I'm uploading core and executable.

(**[file #30718](https://savannah.nongnu.org/support/download.php?file_id=30718)**)

avrdude-admin, 2021-12-10:
Guyzmo <guyzmo>
Thu 27 Feb 2014 02:28:36 PM UTC

Actually, I still don't know how to make that bug reproduceable, but here is some more context:

It's working well for sometime, I upload using the JTAG ICE 3 on two projects (targets being Atmega32u4 and Atmega2560 using ICSP) (as a broad guess, I'm doing about 20/30 uploads a day).

It always work perfectly fine for the Atmega32u4 target, then at some point it fails to work with the Atmega2560 target. And there's nothing I can do to make it work again, though the Atmega32u4 gets still flashed perfectly fine.

I don't remember whether once it begins to crash, I can run avrdude -t or not. I'll have to check that next time this happen.

So please, given the core file and the executable, It'd be really cool to find out what's wrong and point me what would be the useful next steps next time this happen. Sadly, I usually can't keep the flasher in a non-working state for long, as I need it to get on with my work.

N.B.: From time to time, after lots of resets, the atmega32u4 make Xorg crash (I think it's because Xorg sees it as a Mouse, which I still fail to prevent), but I don't think both bugs are related.

avrdude-admin, 2021-12-10:
Guyzmo <guyzmo>
Thu 27 Feb 2014 02:36:28 PM UTC

the core file is only 4MB, but savannah refuses to take it so I posted it over there:

 [http://m0g.net/avrdude.core](https://savannah.nongnu.orghttp://m0g.net/avrdude.core)

(and I tried 4 times, and savannah give me no errors!)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 27 Feb 2014 02:52:42 PM UTC

Trying to see whether it will accept the compressed coredump.

(**[file #30719](https://savannah.nongnu.org/support/download.php?file_id=30719)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Wed 12 Mar 2014 09:00:11 PM UTC

This appears to be a problem when connecting the JTAGICE3 (with
the pre-EDBG firmware) through USB 1.1.  Previously, this
didn't work at all, and there are probably also bugs in
AVRDUDE even for firmware versions where it appears to work.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 21 Feb 2016 09:42:50 PM UTC

Please try AVRDUDE 6.3.  It implements fragmentation.

mcuee, 2022-05-29:
No new reports here. Not so sure if this is still relevant.

MCUdude, 2022-05-29:
I'm closing this issue. The issue appeared to affect Avrdude 6.0.1 and was not easily reproducible. Feel free to open a new issue if this turns out to be an issue with Avrdude 7.0 as well.

closed 2022-05-29 by MCUdude