avrdudes/avrdude #331: [bug #41631] TPI bitbang for preprogrammed ATtinys can fail (Patch attached)

avrdude-admin, 2021-12-10:
Matthias Neeracher <None>
Tue 18 Feb 2014 01:13:56 PM UTC

If an ATtiny4/5/9/10 has been programmed with software that sets PB0 (a.k.a TPIDATA) as an output, this will interfere with the MOSI -> MISO connectivity test when avrdude tries to reprogram the MCU.

The simplest solution, it seems to me, is to hold RESET low during the MOSI->MISO test. I’ve attached a patch implementing this, which works for me (Using a Buspirate as the programmer).

[file #30594: bitbang,patch](https://savannah.nongnu.org/bugs/download.php?file_id=30594)

This issue was migrated from https://savannah.nongnu.org/bugs/?41631

mcuee, 2022-06-02:
Can this be made generic? What does the official Atmel programmers do in this case?

BTW, STM32 Cube Programmer has an option to ask the user to try to enter programming mode while holding the chip under reset. Maybe a document enhancement can be done to provide information for such siutation and potential workaround (if possible).

mcuee, 2022-06-20:
Somehow I can not download the patch now, I will try later.

mcuee, 2022-06-22:
> Somehow I can not download the patch now, I will try later.

Looks like the patch is gone.

MCUdude, 2022-06-22:
I was able to download it!

```patch
Index: bitbang.c
===================================================================
--- bitbang.c	(revision 1275)
+++ bitbang.c	(working copy)
@@ -548,8 +548,8 @@
       return -1;
     }
 
-	/* bring RESET high first */
-    pgm->setpin(pgm, PIN_AVR_RESET, 1);
+	/* bring RESET low first */
+    pgm->setpin(pgm, PIN_AVR_RESET, 0);
 	usleep(1000);
 
     if (verbose >= 2)
@@ -571,6 +571,8 @@
   }
 
   pgm->setpin(pgm, PIN_AVR_SCK, 0);
+  pgm->setpin(pgm, PIN_AVR_RESET, 1);
+  usleep(1000);
   pgm->setpin(pgm, PIN_AVR_RESET, 0);
   usleep(20000);
 

```

mcuee, 2022-10-23:
I do not have TPI parts for testing as of now. Just wondering if this is still applicable or not.

mcuee, 2023-04-09:
@MCUdude 

I do not have good testing facility for TPI parts. Just wondering if you can check if this is still an issue or not. Thanks.

mcuee, 2023-06-05:
Nobody has complained for so long, so I guess this is no longer an issue. I will close this for now. We can re-open this if there are new reports.

closed 2023-06-05 by mcuee