avrdudes/avrdude #332: [bug #41637] avrdude timeout should be specifiable

avrdude-admin, 2021-12-10:
laci@monda.hu <None>
Wed 19 Feb 2014 10:19:05 AM UTC

Currently avrdude timeouts after about 3 seconds when not being able to program the AVR.  I'd like to change this value to about 1 second (with a command line option, without recompiling avrdude).

The reason is that I'm programming an AVR through another AVR that can act like an USB to Serial bridge.  I've written a script to try to program the remote AVR with avrdude and if it doesn't succeed then kick the AVR into bootloader mode.  Being able to specify the timeout would make my script significantly faster.

This issue was migrated from https://savannah.nongnu.org/bugs/?41637

mcuee, 2022-06-02:
This sounds like a niche use case but may still be kind of useful.

MCUdude, 2022-07-29:
>  Currently avrdude timeouts after about 3 seconds when not being able to program the AVR.  I'd like to change this value to about 1 second (with a command line option, without recompiling avrdude).

OP didn't specify which programmer he or she is using? My AVRISPmkII exits immedeatly if no target is detected. However, `-c arduino` takes several seconds before throwing `avrdude: stk500_recv(): programmer is not responding`. I'll assume OP is using `-c arduino`?

MCUdude, 2022-07-29:
Currently, there's no way to interactively change the serial timeout delay specified in ser_posix.c and ser_win32.c.

https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/ser_posix.c#L53

https://github.com/avrdudes/avrdude/blob/affe4cb50a42c34a873be9bde8b5e650032cdcb5/src/ser_win32.c#L39

mcuee, 2023-06-12:
Okay, I will close this now.