avrdudes/avrdude #333: [bug #41671] Feature Request: reinstall erase/rewrite cycle counter

avrdude-admin, 2021-12-10:
Michael H. <de.bug@safe-mail.net>
Sat 22 Feb 2014 07:03:53 PM UTC

Feature Request: reinstall erase/rewrite cycle counter (options -y / -Y). There are actually some users that like to use this funtion. Me being one of them.

This issue was migrated from https://savannah.nongnu.org/bugs/?41671

mcuee, 2022-06-02:
Just wondering how this can be done? Does the AVR chip has such function to know the number of erase/rewrite cycle?

MCUdude, 2022-06-02:
> Does the AVR chip has such function to know the number of erase/rewrite cycle?

Nope, it does not. It simply reads an EEPROM address and increases the number in this memory location every time you erase the chip. I don't really see the real need for this feature...

closed 2022-06-02 by MCUdude