avrdudes/avrdude #337: [bug #41878] AVR Dragon JTAG: ATxmega256A3BU hangs, when programming or reading device signature

avrdude-admin, 2021-12-10:
Andrei <senjas@yandex.ru>
Sun 16 Mar 2014 01:21:30 PM UTC

When reading device signature by issuing
avrdude -px256a3bu -cdragon_jtag
or programming, ATxmega256A3BU hangs and does not execute the written code.

If I reset the chip after accessing it through AVRDUDE,
ATxmega256A3BU works without any problems.

I use AVRDUDE 6.1 on Linux machine (openSUSE 12.3).
Feel free to ask for more details, as I am not sure what info should I provide.

I have also Windows 7 machine with Atmel Studio 6.1.
There I can program ATxmega256A3BU through the same AVR Dragon normally:
the code gets executed right after finishing programming.

This issue was migrated from https://savannah.nongnu.org/bugs/?41878

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 23 Nov 2014 08:50:34 PM UTC

Can you please run with -vvvv (four options "-v"), and
attach the stderr output as a file to this bug report?

mcuee, 2022-06-22:
Not so sure if this is linked to #365 
* #365

mcuee, 2022-07-12:
This may be related to #310.
* #310

MCUdude, 2022-09-23:
I'm not able to reproduce this issue, so I think we can close it.

```
$ ./avrdude -cdragon_jtag -p atxmega256a3bu -Uflash:w:/Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading input file /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex for flash
avrdude: writing 35062 bytes flash ...

Writing | ################################################## | 100% 9.14s

avrdude: 35062 bytes of flash written
avrdude: verifying flash memory against /Users/hans/Downloads/XMEGA_A3BU_XPLAINED_DEMO1.hex

Reading | ################################################## | 100% 7.10s

avrdude: 35062 bytes of flash verified

avrdude done.  Thank you.
```

```
$ ./avrdude -cdragon_jtag -p atxmega256a3bu -Usignature:r:-:h

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9843 (probably x256a3bu)
avrdude: NOTE: Programmer supports page erase for Xmega devices.
         Each page will be erased before programming it, but no chip erase is performed.
         To disable page erases, specify the -D option; for a chip-erase, use the -e option.
avrdude: reading signature memory ...

Reading | ################################################## | 100% 0.01s

avrdude: writing output file <stdout>
0x1e,0x98,0x43

avrdude done.  Thank you.

```

closed 2022-09-23 by MCUdude