avrdudes/avrdude #338: [bug #41928] Avrdude fails to burn flash of Atmega328p using buspirate with Arduino IDE

avrdude-admin, 2021-12-10:
Roberto Aragón <robarago@gmail.com>
Fri 21 Mar 2014 08:37:18 PM UTC

If using the buspirate support on avrdude 6.1 (and 5.5 also tested), the flash gets corrupted the second time it is used with arduino IDE (the first time always ends with success).

The only workaround is to use avrdude with the "-e" switch (or without "-D" to avoid automatic flash erasure). The MCU is then correctly flashed every time, but the RESET support is gone. This is the expected behaviour, since erasing the flash also erases the bootloader and the RESET support.

The short error message shown is something like this:

Attempting to initiate BusPirate binary mode...
avrdude: verification error, first mismatch at byte 0x010a
0x68 != 0x6a
avrdude: verification error; content mismatch

I compiled avrdude 6.1 from scratch in a GNU/Linux machine and copied the configuration file and the executable to the tools dir of the Arduino IDE with identical results.

Maybe this is the expected behaviour. If so, I apologize, I'm a beginner and maybe I misunderstood something about this tool.

This issue was migrated from https://savannah.nongnu.org/bugs/?41928

avrdude-admin, 2021-12-10:
Matthijs Kooijman <matthijs>
Fri 02 May 2014 07:45:18 AM UTC

It would probably help for debugging if you posted the .hex file used and the exact avrdude commandline that the Arduino IDE calls (enable verbose uploads for that).

avrdude-admin, 2021-12-10:
Matthijs Kooijman <matthijs>
Tue 14 Oct 2014 06:57:00 PM UTC

How did you add support for the buspirate to the Arduino IDE? I don't think it's supported out of the box. In Arduino version 1.5.x, the -D option is not passed by default to external programmers defined in `programmers.txt`, so it's likely that this is a problem in the way you added support for the buspirate to the IDE (or perhaps a problem in the IDE, if you use version 1.0.x, which hardcodes more paths and options IIRC).

In any case, I think that running avrdude with -D and without -e is intended to **not** do an erase, meaning you can only change bits from 1 to 0, not vice versa. If you're trying to flash in a new program, this is expected to break...

mcuee, 2022-05-24:
Buspirate has been added.
https://github.com/avrdudes/avrdude/blob/main/src/buspirate.c

And judging from the reply, this can most likely be closed.

MCUdude, 2022-05-29:
I can confirm that the Bus pirate works perfectly fine with Avrdude 7.0:

```
$ ./avrdude -c buspirate -p atmega328p -P /dev/cu.usbmodem000000011 -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_88416/Blink.ino.hex:i

Attempting to initiate BusPirate binary mode...
avrdude: Paged flash write enabled.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_88416/Blink.ino.hex"
avrdude: writing flash (1030 bytes):

Writing | ################################################## | 100% 1.54s

avrdude: 1030 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_88416/Blink.ino.hex:

Reading | ################################################## | 100% 1.37s

avrdude: 1030 bytes of flash verified

avrdude done.  Thank you.
```

closed 2022-05-29 by MCUdude