avrdudes/avrdude #340: [bug #42068] New programmer entry

avrdude-admin, 2021-12-10:
Dmitry Marakasov <amdmi3@amdmi3.ru>
Mon 07 Apr 2014 05:54:07 PM UTC

New programmer entry for simple handmade parallel port programmer widely used by Russian AVR enthusiasts known as "five wires" ("пять проводков"):

Mentions and schematics (Russian, but see images):

[http://www.123avr.com/07.htm](https://savannah.nongnu.orghttp://www.123avr.com/07.htm)
 [http://www.getchip.net/posts/delaem-lpt-programmator-dlya-avr-mikrokontrollerov/](https://savannah.nongnu.orghttp://www.getchip.net/posts/delaem-lpt-programmator-dlya-avr-mikrokontrollerov/)
 [https://www.google.ru/search?q](https://savannah.nongnu.orghttps://www.google.ru/search?q)=пять+проводков&tbm=isch

Name translation:

[http://translate.google.com/#ru/en/пять%20](https://savannah.nongnu.orghttp://translate.google.com/#ru/en/пять%20)проводков

programmer
id    = "five-wires";
desc  = "\"Five wires\" parallel programmer";
type  = par;
sck   = 6;
mosi  = 7;
reset = 9;
miso  = 10;
;

This issue was migrated from https://savannah.nongnu.org/bugs/?42068

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Fri 14 Nov 2014 12:00:16 PM UTC

Hi,

it seems that at [http://avr.ru/ready/tools/prog/5wire](https://savannah.nongnu.orghttp://avr.ru/ready/tools/prog/5wire) the same name is used for a programmer which uses different pins!

So I think it is not a good idea to add it when users think of different hardware when using it.

It is probably better to inform the users on these web pages mentioned below that "stk200" could be used (as its buff pins 4,5 are not used by the five-wires).

Another solution would be to provide there a specific conf for five-wires, which can be added to the existing conf at the command line:

avrdude -C+five-wires.conf ...

(This -C+ is new in avrdude 6, and appends the conf file to the previous read system/user conf.)

BTW: the correct syntax for avrdude 6 is:

programmer
id = "five-wires";
desc = "\"Five wires\" parallel programmer";
type = "par";                <<<<<<<< is now a string
connection_type = parallel;  <<<<<<<< is a new option
sck = 6;
mosi = 7;
reset = 9;
miso = 10;
;

mariusgreuel, 2022-01-12:
Parallel port programmers are no longer supported.