avrdudes/avrdude #341: [bug #42158] Linux GPIO - Source Typo

avrdude-admin, 2021-12-10:
Andy Simpkins <None>
Mon 21 Apr 2014 04:50:27 PM UTC

Hi there,

This is causing me problems with implementing bit
bashed sysfs gpio on armhf (in this case a beagle bone black and
derivative) i.e. type = "linuxgpio"
attempting to configure GPIO line 45 failed with the error message:
"pin must be in the range 0-31"

Many thanks for all your hard work.

/Andy

THE FIX:
The definition HAVE_LINUX_GPIO is used to set PIN_MAX between either 31
(default) or 255, however it is called HAVE_LINUXGPIO in the rest of the
codebase.

pindefs.h:62
#ifdef HAVE_LINUX_GPIO
becomes
#ifdef HAVE_LINUXGPIO

TO REPRODUCE:
I use a custom config file: lwb-avrdude.conf
programmer
id    = "keyboard";
desc  = "Toby Churchill Ltd.  LWB platform, keyboard";
type  = "linuxgpio";
reset = ~45;
sck   = 2;
mosi  = 3;
miso  = 5;
;

call avrdude thus:
avrdude -v -p m32 -C ./lwb-avrdude.conf -c keyboard -e

returns:
avrdude: Version 6.1, compiled on Apr 16 2014 at 17:41:25
Copyright (c) 2000-2005 Brian Dean, [http://www.bdmicro.com/](https://savannah.nongnu.orghttp://www.bdmicro.com/)
Copyright (c) 2007-2014 Joerg Wunsch
System wide configuration file is "./lwb-avrdude.conf"
avrdude: error at line 5 of ./lwb-avrdude.conf: pin must be in the range
0-31

This issue was migrated from https://savannah.nongnu.org/bugs/?42158

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Wed 07 May 2014 08:54:27 PM UTC

Fixed in revision 1299

avrdude-admin, 2021-12-10:
Bas Wijnen <shevek>
Tue 28 Apr 2015 02:00:03 AM UTC

I can confirm that this fix works.  It also is important especially for the BeagleBone, because its GPIOs are overloaded with other functions, so not all of them are available, and in fact almost none of the pins below 32 are available.

So please can this be included in a release some time soon?

Thank you very much,
Bas

MCUdude, 2022-04-04:
I'm closing this issue because the main issue has been resolved. The issue mentioned by Bas Wijnen regarding GIPO pin number has also been resolved.

https://github.com/avrdudes/avrdude/blob/e069871c8e4356c3b742d53903c8504b0de356e3/src/libavrdude.h#L366-L375

closed 2022-04-04 by MCUdude