avrdudes/avrdude #343: [bug #42310] New part description for AT90PWM216

avrdude-admin, 2021-12-10:
cnobile, skotti <None>
Tue 06 May 2014 10:19:53 PM UTC

The AT90PWM216 can easily added to the avrdude.conf file by referencing to the AT90PWM316.

See also
 [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1156804](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1156804)
and
 [https://savannah.nongnu.org/bugs/?21797](https://savannah.nongnu.orghttps://savannah.nongnu.org/bugs/?21797)

```
#------------------------------------------------------------
# AT90PWM216
#------------------------------------------------------------
# Completely identical to AT90PWM316 (including the signature!)

part parent "pwm316"
id = "pwm216";
desc = "AT90PWM216";
;

```

This issue was migrated from https://savannah.nongnu.org/bugs/?42310

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Wed 07 May 2014 09:20:28 PM UTC

added in revision 1300

MCUdude, 2022-01-09:
Part is present in the current avrdude.conf file. Issue can be closed

closed 2022-01-09 by mariusgreuel