avrdudes/avrdude #344: [bug #42324] Feature Request: Refactor rts/dtr toggling as generic option

avrdude-admin, 2021-12-10:
David Zanetti <dave2@wetstring.net>
Fri 09 May 2014 09:43:42 AM UTC

Currently two different programmer types (arduino and wiring) each individually implement RTS/DTS toggling for automatic reset into a bootloader on a target board.

I'd like this to be a generic option that could be applied to any serial-based programmer, either as a command-line option or programmer config feature.

It would be useful for bootloaders like xboot (
 [http://alexforencich.com/wiki/en/xboot/start](https://savannah.nongnu.orghttp://alexforencich.com/wiki/en/xboot/start)) which implement avr109 as a protocol. Currently I've patched butterfly.c to do auto-reset, but it seems more sensible to make it a generic feature.

Thanks!

This issue was migrated from https://savannah.nongnu.org/bugs/?42324

avrdude-admin, 2021-12-10:
Didrik Madheden <didrikm>
Sat 10 May 2014 05:13:10 AM UTC

I concur. This ought to be configurable in a similar way to the serbb programmer, with a reset and (why also not) a power statement that can be mapped individually to dtr and rts to allow either signal and either polarity to control the reset state.

/Didrik

mcuee, 2022-06-06:
This seems to be a nice feature if it can be done.

mcuee, 2022-06-06:
Similar discussion:
* #980 
* https://github.com/avrdudes/avrdude/issues/788

mcuee, 2023-01-07:
Issue #1262 is more or less the same as this one.

mcuee, 2023-01-07:
I will close this one and further discussion can be done in #1262.

closed 2023-01-07 by mcuee