avrdudes/avrdude #348: [bug #42517] "configure" script of avrdude-6.1 does not detect availability of some header files

avrdude-admin, 2021-12-10:
Jürgen Harms <None>
Fri 06 Jun 2014 02:18:57 PM UTC

Trying to build/install avrdude-6.1 fails (at least on Mageia) due to a missing definition of the type uint8_t.

The reason for this is that the "configure" script detects the <stdint.h> header file as not available and, consequently, does not define the HAVE_STDINT_H variable - although the file /usr/include/stdint.h exists in the distro.

Fix: the script (line #5650 of the "configure" file in the 6.1 tarball) contains a list of header files to be checked - stdint.h is missing in that list; to fix the bug, simply add stdint.h to this list.

Verifying the file ac_cfg.h, it appears that - in addtion to <stdint.h> - there are several other header files that likewise are not detected.

The patch file provided as an attachment corrects the "configure" script to iterate over all files for which HAVE_<xxx>_H variables are enumerated in ac_cfg.h. This patch has been successfully tested: avrdude now builds and executes without problem.

[file #31515: avrdude-6.1_includes.patch](https://savannah.nongnu.org/bugs/download.php?file_id=31515)

This issue was migrated from https://savannah.nongnu.org/bugs/?42517

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Sat 21 Jun 2014 10:10:13 PM UTC

For the current build/bootstrap process these header files are included (rev 1324.) Probably one of the latest changes (libtool) fixed it.

avrdude-admin, 2021-12-10:
Jürgen Harms <harms>
Sun 22 Jun 2014 03:22:01 PM UTC

> Probably one of the latest changes (libtool) fixed it.

Or there might be a platform- (or library version-) dependant problem. When I first (before posting this as a bug) discussed this on the avrdude ML, I was struck by the fact that several responses were "works for me" - on my platform the problem was solid.

I will need to generate a package for Mageia-5 Alpha (mid-july) with the latest libraries (presently using what is canned into Mageia-4) - and, as required by Mageia packaging policies, will need to use the latest Avrdude tarball published as stable Will add a comment once this is done.

avrdude-admin, 2021-12-10:
Jürgen Harms <harms>
Thu 10 Jul 2014 10:45:22 AM UTC

I have now re-tried on Mageia-5 (Alpha-1): the problem persists unless I apply the patch submitted before I configure and make avrdude from the tarball. Since I do this in preparation of building an rpm package, Mageia policy makes me stick to the tarball available as a stable version (12-Mar-2014) - plus eventually some specific patches - hence, I can live with the present situation.

OT: Building the rpm package under Mageia-5 (as opposed to doing this on Mageia-4) there is a new problem: the build process implies running automake. Mageia-5 comes with automake-1.14.1 and the build fails complaining about an incompatibility with code for automake-1.14 - it appears that configure.ac of the tarball is specific to automake-1.14. I have not yet resolved this problem.

MCUdude, 2022-01-09:
Something tells me that this isn't a problem anymore...

mariusgreuel, 2022-01-09:
Works for both automake & cmake.

closed 2022-01-09 by mariusgreuel