avrdudes/avrdude #352: [bug #43078] AVRDUDE crashes after sucessfully reading/writing eeprom

avrdude-admin, 2021-12-10:
Janne Virsunen <janne.virsunen@syncore.se>
Tue 26 Aug 2014 08:38:50 AM UTC

Using version 6.1 of AVRDUDE, with an additional custom part "m6490p" (included avrdude.conf, line 6000-6012), the program crashes after a successful read or write of the eeprom and thusly does not return '0' as expected after a successful execution.

Operating system used is Windows 7.
Log file of output from AVRDUDE included, additional information from the crash box that appears in Windows aswell as the eeprom.eep file that is written into the eeprom of the m6490p.

Drivers used for jtag3-programmer is libusb0 (v1.2.6.0)

[file #31962: log.txt](https://savannah.nongnu.org/bugs/download.php?file_id=31962)
[file #31965: crashbox.txt](https://savannah.nongnu.org/bugs/download.php?file_id=31965)
[file #31964: avrdude.conf](https://savannah.nongnu.org/bugs/download.php?file_id=31964)
[file #31963: eeprom.eep](https://savannah.nongnu.org/bugs/download.php?file_id=31963)
[file #32661: main.eep](https://savannah.nongnu.org/bugs/download.php?file_id=32661)

This issue was migrated from https://savannah.nongnu.org/bugs/?43078

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Sun 23 Nov 2014 08:48:39 PM UTC

I tracked down an issue here that was also reported by
valgrind as resulting in heap corruption.  I think this
would also fix your bug.

If you have the chance to test the SVN version r1344, please
do so, and report whether your problem has been solved.

avrdude-admin, 2021-12-10:
Mihai Nedelcu <mnedix>
Sat 13 Dec 2014 10:01:30 PM UTC

Hello,
I have the same issue with avrdude crashing when trying to program the .eep file. When crashing, it returns the following:
Detected 1e960a = ATmega644P
~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~ ~

avrdude.exe: set SCK frequency to 1500000 Hz
avrdude.exe: warning: cannot set sck period. please check for usbasp firmware update.
avrdude.exe: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude.exe: Device signature = 0x1e960a
avrdude.exe: reading input file "D:\Electronics\Electronics\Nixie\Spectrum\_\main.eep"
avrdude.exe: input file D:\Electronics\Electronics\Nixie\Spectrum\_\main.eep auto detected as Intel Hex
avrdude.exe: writing eeprom (0 bytes):

Writing | ################################################## | 100% -0.00s

avrdude.exe: 0 bytes of eeprom written
avrdude.exe: verifying eeprom memory against D:\Electronics\Electronics\Nixie\Spectrum\_\main.eep:
avrdude.exe: load data eeprom data from input file D:\Electronics\Electronics\Nixie\Spectrum\_\main.eep:
avrdude.exe: input file D:\Electronics\Electronics\Nixie\Spectrum\_\main.eep auto detected as Intel Hex
avrdude.exe: input file D:\Electronics\Electronics\Nixie\Spectrum\_\main.eep contains 0 bytes
avrdude.exe: reading on-chip eeprom data:

Reading |

...and Windows is popping up the crash window.

I see in the previous response that another version of avrdude may fix the issue but I downloaded all releases and they all behave the same. Is there any solution on this?

(**[file #32661](https://savannah.nongnu.org/support/download.php?file_id=32661)**)

avrdude-admin, 2021-12-10:
Mihai Nedelcu <mnedix>
Sun 14 Dec 2014 09:18:56 AM UTC

Hello, problem solved. Turned out the .eep file was empty and that's why it could not be programmed.
As a suggestion, perhaps an error message can be included in avrdude, something like "File empty, programming not possible."

avrdude-admin, 2021-12-10:
Rene Liebscher <rliebscher>
Sun 14 Dec 2014 10:48:37 AM UTC

Empty eeprom file crashes were fixed for **[bug #40142](https://savannah.nongnu.org/bugs/?40142)**.
However a new release of avrdude was not done since then.

MCUdude, 2022-01-07:
Issue can be closed

closed 2022-01-08 by dl8dtl