avrdudes/avrdude #354: [bug #43178] Problem programmig uc3a0512 with AtmelIce, target does not answer

avrdude-admin, 2021-12-10:
Bradley Jarvis <brad@pocketinnovations.com.au>
Tue 09 Sep 2014 01:33:26 PM UTC

I am trying to program an uc3a0512 AVR32 using an Atmel Ice programming tool. I have svn 1333 release of avrdude running on gentoo. I am able to program an ATxmega32e5 using the PDI interface with no problem, however the JTAG interface to the uc3a0512 does not work.I have used the programmer with Atmel Studio 6.2 and at32program with no problem.

I have attached a log of an upload attempt, the command line run was

avrdude -vvvv -p uc3a0512 -c atmelice -B 1 -Uflash:w:../firmware/fw.rom -l avrdude-at32uc3a0512-program.log

Any insight as to what the problem may be would be gratefully received.

[file #32048: avrdude-at32uc3a0512-program.log](https://savannah.nongnu.org/bugs/download.php?file_id=32048)
[file #32055: avrdude-jtagmkii-uc3a0512-program.log](https://savannah.nongnu.org/bugs/download.php?file_id=32055)

This issue was migrated from https://savannah.nongnu.org/bugs/?43178

avrdude-admin, 2021-12-10:
Bradley Jarvis <bradley_jarvis>
Thu 11 Sep 2014 05:43:07 AM UTC

Just to add some extra information from my original submission. I also have a jtagmkii and have found that this is also not able to program the uc3a0512. I have tried both jtagmkii and jtagmkii_avr32. I have included a log of both these program attempts. I have noticed that the avr32 version certainly gets a bit further along in being able to program the device. I did notice in somewhere that someone mentioned that the EDBG usb interface is just a wrapper around the jtag interface, if this is the case then should there be a atmelice_avr32 option?

Looking at the log for the jtagmkii_avr32 it is fine up until it tries to write the firmware where it starts to get error 0xA4. The jtagmkii option fails with Illegal JTAG ID.

As with the atmelice I have tested the jtag mk2 in avrstudio 6.2 with no problem. I tried looking at the atmelice usb packets when programming from atmel studio 6.2 but it continually streams data to the device making it very hard to see what is happening. I would be happy to provide this information if required.

(**[file #32055](https://savannah.nongnu.org/support/download.php?file_id=32055)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Thu 11 Sep 2014 07:23:43 AM UTC

Alas, I am afraid there is no active developer who is working,
or even has the ability to test on UC3 devices.

If you want to compare what AVRDUDE does against the actions
of Atmel Studio, and can suggest what we need to change, you
are welcome.

avrdude-admin, 2021-12-10:
Bradley Jarvis <bradley_jarvis>
Mon 15 Sep 2014 02:01:11 AM UTC

Hi Joerg,
Ok I have now got avrdude downloading to the AVR32, I have ended up added a new programmer called atmelice_avr32. The update includes mostly changes to jtag3.c and jtag3_private.h, i also updated pgm_type.c and avrdude.conf to include the "new" programming device. At the moment the change is a bit messy, I am planning on cleaning it up once I get some free time (I also might look into why the jtagmkii_avr32 functions failed).

The jtag interface to the avr32 ended up being pretty different to the avr8's so I have implemented extra functions to handle this stuff. At the moment I added

const char jtag3_avr32_desc[] = "Atmel JTAGICE3 in AVR32 mode";
void jtag3_avr32_initpgm(PROGRAMMER * pgm)

Which include handles to new avr32 specific functions, it is possibly cleaner though to add a avr32 flag and use the avr8 functions with a check to the flag to select between code, what do you think?

Do you want me to upload a patch for my current code so you can suggest what you would like for this to be accepted into avrdude?

I am also happy to maintain the avr32 specific code too.

Regards Brad

avrdude-admin, 2021-12-10:
Al <anonuc3user>
Fri 18 Mar 2016 01:37:19 PM UTC

I'll just add to the number of ppl interested in UC3 support, and report that I have experienced the excact same behaviour as described by Bradley Jarvis with the atmelice programmer. The atmelice_avr32 programmer is not listed, so I assume it hasn't been added to the project yet.

mcuee, 2022-05-24:
The patch which supposedly fix the issue is here.
https://github.com/avrdudes/avrdude/issues/735

mcuee, 2022-07-22:
Since we closed #735, we will close this as well -- AVR32 support will not be actively worked on.

closed 2022-07-22 by mcuee