avrdudes/avrdude #356: [bug #43268] usb_drain() call causes LUFA AVR-ISP MKII Code to Fail

avrdude-admin, 2021-12-10:
Colin O'Flynn <None>
Sun 21 Sep 2014 11:31:24 AM UTC

I had seen this reported on AVRFreaks (originally at [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=819522](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=819522) but I don't know the new link).

The LUFA AVR-ISP MKII has different endpoint settings than a 'real' AVR-ISP MKII, due to some compatibility issues with libusb. This seems to be triggered by the call to drain() on the USB endpoint.

Easy fixes I see:

1) Turn it off on a per-programmer basis (i.e. for AVR-ISP MK2 only), probably via the config file.

2) Turn if off globally, but with a command-line switch to re-enable it in case it broke something

Opening this bug report to track the issue.

[file #32159: RZRAVEN-AVRISP-MKII.hex](https://savannah.nongnu.org/bugs/download.php?file_id=32159)
[file #32160: lufa-avrisp2.xml.bz2](https://savannah.nongnu.org/bugs/download.php?file_id=32160)
[file #32161: atmel-avrisp2.xml.bz2](https://savannah.nongnu.org/bugs/download.php?file_id=32161)

This issue was migrated from https://savannah.nongnu.org/bugs/?43268

avrdude-admin, 2021-12-10:
Colin O  Flynn <c_oflynn>
Sun 21 Sep 2014 11:38:40 AM UTC

I now notice that in usb_libusb.c there is a separate struct for the AVR-ISPMK2, which might make it possible to disable the drain for that device without affecting anything else. Need to confirm that struct isn't used by other programmers.

avrdude-admin, 2021-12-10:
Colin O  Flynn <c_oflynn>
Mon 22 Sep 2014 01:59:49 AM UTC

Have now added a hex-file for testing with RZRAVEN USB Stick. There is no IO pins connected, so you can't program a real AVR. I don't even know where the IO pins are routed. It also doesn't route the LEDs or anything so they don't turn on. You can basically just confirm with dmesg it appears as expected.

You can run a command with -F to fake a read, but it will always be garbage.

I've confirmed the hex file does work as expected. Although now on one of my systems it works as expected (almost, just errors out once). Although on another computer you get this, which is the bug being triggered:

[root@localhost colin]# avrdude -P usb -p m328p -c avrisp2

avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_recv_mk2: error in USB receive
avrdude: stk500v2_recv_mk2: error in USB receive
^C

i.e. you get that 'error in USB receive' in a loop and nothing else. It's based on the latest release of LUFA (140302).

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 22 Sep 2014 08:45:29 AM UTC

OK, using it under Linux, it "just works":

```
% avrdude -P usb -p m328p -c avrisp2 -vvvv

avrdude: Version 6.1-20140519, compiled on May 19 2014 at 12:02:47
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "/usr/local/etc/avrdude.conf"
User configuration file is "/home/jwunsch/.avrduderc"
User configuration file does not exist or is not a regular file, skipping

Using Port                    : usb
Using Programmer              : avrisp2
avrdude: usbdev_open(): Found AVRISP mkII, serno: 000200212345
avrdude: Sent: . [01]
avrdude: Recv: . [01] . [00] . [0a] A [41] V [56] R [52] I [49] S [53] P [50] _ [5f] M [4d] K [4b] 2 [32]
avrdude: stk500v2_getsync(): found AVRISP mkII programmer
AVR Part                      : ATmega328P
Chip Erase delay              : 9000 us
PAGEL                         : PD7
BS2                           : PC2
RESET disposition             : dedicated
RETRY pulse                   : SCK
serial program mode           : yes
parallel program mode         : yes
Timeout                       : 200
StabDelay                     : 100
CmdexeDelay                   : 25
SyncLoops                     : 32
ByteDelay                     : 0
PollIndex                     : 3
PollValue                     : 0x53
Memory Detail                 :

Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
Block Poll               Page                       Polled
Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

Programmer Type : STK500V2
Description     : Atmel AVR ISP mkII
Programmer Model: AVRISP mkII
avrdude: Sent: . [03] . [90]
avrdude: Recv: . [03] . [00] . [00]
avrdude: Sent: . [03] . [91]
avrdude: Recv: . [03] . [00] . [01]
avrdude: Sent: . [03] . [92]
avrdude: Recv: . [03] . [00] . [17]
Hardware Version: 0
Firmware Version Master : 1.23
avrdude: Sent: . [03] . [94]
avrdude: Recv: . [03] . [00] . [1f]
Vtarget         : 3.1 V
avrdude: Sent: . [03] . [98]
avrdude: Recv: . [03] . [00] . [12]
SCK period      : 19.37 us

avrdude: Sent: . [10] . [c8] d [64] . [19]   [20] . [00] S [53] . [03] . [ac] S [53] . [00] . [00]
avrdude: Recv: . [10] . [c0]
avrdude: stk500v2_command(): command failed
avrdude: Sent: . [03] . [a1]
avrdude: Recv: . [03] . [00] . [10]
avrdude: stk500v2_program_enable(): bad AVRISPmkII connection status: Target not detected
avrdude: initialization failed, rc=-1
Double check connections and try again, or use -F to override
this check.

avrdude: Sent: . [11] . [01] . [01]
avrdude: Recv: . [11] . [00]

avrdude done.  Thank you.

```
I think that's basically the expected behaviour for the fake
programmer.  Under Windows (in a VM), I can see the timeouts
though.

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 22 Sep 2014 09:45:32 AM UTC

OK, fired the USB analyzer onto it.

The difference between Linux and Windows from the analyzer's
point of view is that Windows transactions are wrapped in a
countainer declaring them as "BULK IN" (or "BULK OUT"), Linux
transactions aren't.

Then, inside the Windows "BULK IN", the incomplete transaction
is flagged as "Malformed", commented as "The handshake is
missing".  In contrast, the incomplete Linux transactions are
just that, incomplete only (which is expected since no data
are present).

Strange enough, if I connect an Atmel AVRISPmkII, there is no
sign at all from the usb_drain() visible on the bus. I don't
understand why they are handled differently though.

I'm attaching the analyzer trace reports for anyone who is
interested.  In the lufa-avrisp2.xml file, the interesting
spot is at time 2.074379450 (Linux usb_drain()), and time
12.669058217 (Windows usb_drain()).  In atmel-avrisp2.xml,
the respective OUT transaction (CMD_SIGN_ON) is at time
10.247395683.

(**[file #32160](https://savannah.nongnu.org/support/download.php?file_id=32160)**, **[file #32161](https://savannah.nongnu.org/support/download.php?file_id=32161)**)

avrdude-admin, 2021-12-10:
Joerg Wunsch <joerg_wunsch>
Mon 22 Sep 2014 10:25:04 AM UTC

While I still believe it is only masking a more severe issue
buried deep below, I turned usbdev_drain() into a dummy as
I think it serves no useful purpose (other than triggering
the issue buried deep below ;-).

mariusgreuel, 2022-01-12:
Programmer need to update to the latest LUFA firmware that corrects the endpoint addresses.

mariusgreuel, 2022-01-12:
Dup of #321.

closed 2022-01-12 by mariusgreuel